MyRobotLab/InMoov #1: Merge the all.

mayaway, 2017-01-03:
I have followed the instructions here: http://myrobotlab.org/content/inmoov-script-merge-them-all#comment-8033 
I have created BasicConfig.ini 
I have set my Arduino COM3, COM4 Port speeds to 115200 
I run START_INMOOV.bat (I removed the recreation/unpacking/install all)
I get the Swing GUI then Chome and hear, "Starting Right Hand, Starting Mouth, Ok, You have my attention"
ALL VOICE COMMANDS FAIL
Activate Voice
Attach Right Hand
Close your hand.
Close Your Right Hand

The commands are recognized and mirrored as text. There is NO ACTION by the InMoov
Here is my log - any ideas? It would be wonderful to have a simple setup that just worked out of the box. Then we could build on that. More documentation is needed for sure.
-----------------------
08:46:21.171 [main] INFO c.m.s.Runtime [Runtime.java:1207] -fromAgent -invoke python execFile C:\mrl/InmoovScript/Inmoov.py -service GUIService GUIService python Python 
08:46:21.202 [main] INFO c.m.f.Service [Service.java:1006] getMessageSet loading 254 non-sub-routable methods
08:46:21.202 [main] INFO c.m.f.Service [Service.java:1483] cfg file C:\mrl\.myrobotlab\runtime.json does not exist
08:46:21.389 [main] INFO c.m.f.r.ServiceData [ServiceData.java:78] try #1 loading local file C:\mrl\.myrobotlab\serviceData.json
08:46:21.436 [main] INFO c.m.s.Runtime [Runtime.java:1677] ============== args begin ==============
08:46:21.436 [main] INFO c.m.s.Runtime [Runtime.java:1689] jvmArgs [-Djava.library.path=libraries/native, -Djna.library.path=libraries/native]
08:46:21.436 [main] INFO c.m.s.Runtime [Runtime.java:1691] args [-fromAgent, -invoke, python, execFile, C:\mrl/InmoovScript/Inmoov.py, -service, GUIService, GUIService, python, Python]
08:46:21.436 [main] INFO c.m.s.Runtime [Runtime.java:1693] ============== args end ==============
08:46:21.436 [main] INFO c.m.s.Runtime [Runtime.java:1695] ============== env begin ==============
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] configsetroot=C:\windows\ConfigSetRoot
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] USERDOMAIN_ROAMINGPROFILE=Maya
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] LOCALAPPDATA=C:\Users\Mayaway\AppData\Local
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] PROCESSOR_LEVEL=6
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] FP_NO_HOST_CHECK=NO
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] USERDOMAIN=Maya
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] LOGONSERVER=\\MAYA
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] PROMPT=$P$G
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] SESSIONNAME=Console
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] ALLUSERSPROFILE=C:\ProgramData
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] PROCESSOR_ARCHITECTURE=x86
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] PSModulePath=C:\windows\system32\WindowsPowerShell\v1.0\Modules\
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] SystemDrive=C:
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] =ExitCode=00000080
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] =C:=C:\mrl
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] APPDATA=%%CD%%
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] USERNAME=Mayaway
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] CommonProgramFiles=C:\Program Files\Common Files
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] Path=C:\ProgramData\Oracle\Java\javapath;C:\Program Files\Broadcom\Broadcom 802.11 Network Adapter\Driver;;C:\Program Files\Intel\TXE Components\TCS\;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Intel\TXE Components\DAL\;C:\Program Files\Intel\TXE Components\IPT\;C:\Program Files\Lenovo\Bluetooth Software\
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] OS=Windows_NT
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] COMPUTERNAME=MAYA
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] PROCESSOR_REVISION=3703
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] ComSpec=C:\windows\system32\cmd.exe
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] ProgramData=C:\ProgramData
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] HOMEPATH=\Users\Mayaway
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] SystemRoot=C:\windows
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] TEMP=C:\Users\Mayaway\AppData\Local\Temp
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] HOMEDRIVE=C:
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] PROCESSOR_IDENTIFIER=x86 Family 6 Model 55 Stepping 3, GenuineIntel
08:46:21.452 [main] INFO c.m.s.Runtime [Runtime.java:1700] USERPROFILE=C:\Users\Mayaway
08:46:21.468 [main] INFO c.m.s.Runtime [Runtime.java:1700] TMP=C:\Users\Mayaway\AppData\Local\Temp
08:46:21.468 [main] INFO c.m.s.Runtime [Runtime.java:1700] ProgramFiles=C:\Program Files
08:46:21.468 [main] INFO c.m.s.Runtime [Runtime.java:1700] PUBLIC=C:\Users\Public
08:46:21.468 [main] INFO c.m.s.Runtime [Runtime.java:1700] NUMBER_OF_PROCESSORS=4
08:46:21.468 [main] INFO c.m.s.Runtime [Runtime.java:1700] windir=C:\windows
08:46:21.468 [main] INFO c.m.s.Runtime [Runtime.java:1700] =::=::\
08:46:21.468 [main] INFO c.m.s.Runtime [Runtime.java:1702] ============== env end ==============
08:46:21.468 [main] INFO c.m.s.Runtime [Runtime.java:1706] ============== normalized ==============
08:46:21.468 [main] INFO c.m.s.Runtime [Runtime.java:1707] 2017/01/03 08:46:21 - GMT - 2017/01/03 13:46:21
08:46:21.499 [main] INFO c.m.s.Runtime [Runtime.java:1708] Pid 5716
08:46:21.499 [main] INFO c.m.s.Runtime [Runtime.java:1709] ivy [runtime,x86.32.windows]
08:46:21.499 [main] INFO c.m.s.Runtime [Runtime.java:1710] os.name [Windows 8.1] getOS [windows]
08:46:21.499 [main] INFO c.m.s.Runtime [Runtime.java:1711] os.arch [x86] getArch [x86]
08:46:21.499 [main] INFO c.m.s.Runtime [Runtime.java:1712] getBitness [32]
08:46:21.499 [main] INFO c.m.s.Runtime [Runtime.java:1713] java.vm.name [Java HotSpot(TM) Client VM] getVMName [java hotspot(tm) client vm]
08:46:21.499 [main] INFO c.m.s.Runtime [Runtime.java:1714] version [1.0.1867]
[... 660 lines cut ...]
08:47:05.339 [WebGui] INFO c.m.f.Service [Service.java:2027] subscribe [HealthCheck/publishStatus ---> WebGui/onStatus]
08:47:05.343 [WebGui] INFO c.m.f.Service [Service.java:2027] subscribe [HealthCheck/publishState ---> WebGui/onState]
08:47:05.355 [python.interpreter.3] INFO c.m.f.Service [Service.java:2158] HealthCheck info starting clock
08:47:05.359 [python.interpreter.3] INFO c.m.f.Service [WebkitSpeechRecognition.java:123] Start listening event seen.
08:47:05.439 [python.interpreter.3] INFO c.m.s.MarySpeech [MarySpeech.java:86] speakInternal Blocking true Text: ok you have my attention
08:47:05.631 [Thread-25] INFO c.m.s.MarySpeech [MarySpeech.java:225] Starting to speak: ok you have my attention
08:47:05.731 [Thread-25] INFO c.m.s.MarySpeech [MarySpeech.java:61] Ok.. here we go.
08:47:06.393 [New I/O worker #2] INFO c.m.s.WebGui [WebGui.java:496] POST /api/messages
08:47:06.397 [New I/O worker #2] INFO c.m.f.Service [WebkitSpeechRecognition.java:101] Recognized : >call John Locke<
08:47:06.401 [New I/O worker #2] INFO c.m.s.WebGui [WebGui.java:496] POST /api/messages
08:47:06.405 [New I/O worker #2] INFO c.m.f.Service [WebkitSpeechRecognition.java:63] Publish Text : call John Locke
08:47:08.855 [Thread-25] INFO c.m.s.MarySpeech [MarySpeech.java:232] End speaking: ok you have my attention
08:47:08.855 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.
08:47:20.331 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.
08:47:35.331 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.
08:47:50.333 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.
08:48:05.337 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.
08:48:20.339 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.
[... cut ...]
08:49:05.342 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.
08:49:20.344 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.
08:49:35.346 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.
08:49:50.347 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.
08:50:05.351 [python.input] INFO c.m.f.Service [WebkitSpeechRecognition.java:116] Resume listening event seen.

mayaway, 2017-01-03:
With the "Open your finger, Close your finger" commands the software responds as if it has performed the action, but no action is performed.

mayaway, 2017-01-03:
Uh-oh - I see 08:47:00.304 [python.interpreter.3] INFO c.m.f.Service [Arduino.java:1688] setting board to type uno
I have two Mega's -- one on COM3 (left) and COM4 (right)

moz4r, 2017-01-03:
Thank you for this detailed bug report and congrats for the #1 bug open !  time to power on the robot I will test the last wip now about  fingers commands

moz4r, 2017-01-03:
Problem identified in last mrl version, we need to do more test. 
Zip updated with a worky version , mrl 1861 : https://github.com/MyRobotLab/inmoov/archive/master.zip
Can you confirm ?

mayaway, 2017-01-04:
Hi there,

I have downloaded and verified a complete new install of version 1861
My log file is here: 

The scripts correctly verbally reported a communication error with COM4 since my InMoov was NOT attached.
I will now look at config files and give it a go…

Thanks!
Stephen

mayaway, 2017-01-04:
> yehh !!
> ear.addCommand("open your finger", "python", "fingeropen") : call the function :
>
> def fingeropen():
>   i01.moveHand("right",0,0,0,0,0)
>   talkBlocking("ok I open my finger") :
> i01.moveHand("right",0,0,0,0,0):
> 0,0,0,0,0 are the position of 5 fingers
> it's the begining of about gestures , more will be integrated very soon. and chatbot too. and much more.
> we need a 100% clean and worky base before

So now I am trying to add in the left hand, and not having much success.
I have modified:

InMoovSkeleton/leftHand.config
	isLeftHandActivated=1

services/4_Inmoov.py

#Inmoov Left / right
if ScriptType=="RightSide" or ScriptType=="Full":
	right = Runtime.createAndStart("i01.right", "Arduino")
	RightPortIsConnected=CheckArduinos(right,MyRightPort)
	left = Runtime.createAndStart("i01.left", "Arduino")
	LeftPortIsConnected=CheckArduinos(left,MyLeftPort)

InmoovScript/Inmoov.py

def fingeropen():
  i01.moveHand("right",0,0,0,0,0)
  i01.moveHand("left",0,0,0,0,0)
  talkBlocking("ok I open my hand")

def fingerclose():
  i01.moveHand("right",180,180,180,180,180)
  i01.moveHand("left",180,180,180,180,180)
  talkBlocking("ok my hand is closed")

def fingermiddle():
  i01.moveHand("right",90,90,90,90,90)
  i01.moveHand("left",90,90,90,90,90)
  talkBlocking("ok my hand is half closed")

------------
And the starting of the left hand is never announces and it remains dormant.
Any ideas? Where do I i01.start or trigger that for the left hand?

Thanks!
Stephen

moz4r, 2017-01-04:
You have passed fingerstarter test so you need to declare a second arduino 

ScriptType = **Full** into inmoov.config

mayaway, 2017-01-04:
The script says:

Full: Inmoov is Alive ! / not fully implemented yet

Still ok to start with what is there…

Thanks,
S~

closed 2017-01-06 by moz4r