MyRobotLab/InMoov #3: MAP SETTINGS

moz4r, 2017-01-10:
hairygael: the slider in the swing shows min42 max150
hairygael: 0 / 180 is my MINIMUM/MAXIMUM and 42/150 is my MINIMUM_MAP/MAXIMUM_MAP

hairygael, 2017-01-10:
I just did a test inverting my setting values and it seems to work but it is not very logic and should be managed another way, I think.
I have now: 42/150 is my MINIMUM/MAXIMUM and 0/180 is my MINIMUM_MAP/MAXIMUM_MAP
the slider in the swing shows min0 max180, which is good.
The arm is now doing the correct range, which is good also

moz4r, 2017-01-10:
I'm happy if You have a worky servo! I saw discussions about map function and set min/max method that are not correct used. I think more discussions will come about how to standardize those things.

closed 2017-01-13 by moz4r