MyRobotLab/InMoov #17: mappings and min max

hairygael, 2017-01-19:
Hello Anthony,
Mappings and min max are still swapped. The way they currently work will certainly confuse builders.

Normally the min and max should stay as the default values, unless you built a different kind of robot.
And the mappings is where you can adapt/ extend/map your servo range to fit the min and max values, but currently it is set the other way around which doesn't really make sense.

By swapping my values I can make it work but it is not really proper I think.

moz4r, 2017-01-19:
Hi ! arrg map confuse for me too now :)
But I think I know why it is not worky well for you, it is because  There is no setminmax() fonction anymore in the script.
Before I bring it to life again, I need to understand something about map. sometime your min/max are at the begining of the map, sometime at the end. 

Time to bring some lights
I will post here : http://myrobotlab.org/content/servo-mapper.

moz4r, 2017-01-28:
Hi Gael ! Sorry about delay to awmser you. this is the map function : 

map(min_input,max_input,min_output,max_output)

servo physical min/max are set into min_output,max_output
min_input,max_input are used to tweak the movement.

like in your script :  

leftArm.rotate.map(40,180,60,142)

>> 60, 142 are your servo limits values
>> 40,180 : a reference used by gestures

if my servo values are 40 120
so : leftArm.rotate.map(40,180,40,120)

Are we agree about that ?
If I understand, I just need to set as default your input mapings into the config files I think

hairygael, 2017-01-28:
Hello Anthony!

60, 142 are my servo mapping values
40,180 : a reference used as default min max servo limits values

40,180, the servo can go further but if set further than 40 the biceps
collapse with the torso.

here is how it should be by default:

[SERVO_MINIMUM]
bicep=0
shoulder=0
rotate=40
omoplate=10
[SERVO_MAXIMUM]
bicep=90
shoulder=180
rotate=180
omoplate=80
[MINIMUM_MAP]
bicep=0
shoulder=0
rotate=0
omoplate=0
[MAXIMUM_MAP]
bicep=180
shoulder=180
rotate=180
omoplate=180

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-01-28:
mmm not totaly agree :

Simple example based on your values : 
head.jaw.map(0,180,42,101)
42 / 101 are the maximum servo limits about your jaw servo. here we are ok ? I
because map(min_input,max_input,**min_output**,**max_output**)

So about rotate servo this is why I was thinking 60 / 142 are your servo value. And I think there is a strange thing about that

leftArm.rotate.map(40,180,**60**,**142**)

hairygael, 2017-01-28:
Please don't base on InMoovDeep values.
I think it gets everybody confused. The servos are different than the
original 805bb.
Even the jaw servo on Deep is not what I recommand in my BOM.
Jaw servo is supposed to be rotating 0,180, but I have set a HK15298b which
only rotates 0,90.
So to have a min,max that OpenCV accepts for the tracking, I have done
(0,180,42,101) It was (at the time) the only way to fool the PID values.
So the logic configuration was:
minmax(10,25)
map(0,180,42,101)
[image: Images intégrées 1]

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-01-29:
Ok , thank for this information ! this is clear. this is why tracking need to be refactored. And those values was so strange to me. I need to learn too more things about pid + tracking + servo maps

About your initial problem, we need to resolve it

at this time, this is the translation the script does , based on the config file in your post  :
( I think I will rename values like this, more clear )

[MINIMUM_MAP_INPUT] 
rotate=0 
[MAXIMUM_MAP_INPUT] 
rotate=180 
[SERVO_MINIMUM_MAP_OUTPUT] 
rotate=40 
[SERVO_MAXIMUM_MAP_OUTPUT] 
rotate=180 

->> **leftArm.rotate.map(0,180,40,180)**

In gui cursor go from 0 to 180 and servo don't move from 40 to 180

hairygael, 2017-01-29:
That is perfect. Thanks for taking the time on this!
Let me know when you are ready and I will test on both robots to see if it
goes okay.

By the way I have read the article in Planete Robot. You did great!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-02-01:
it is in the box !

hairygael, 2017-02-02:
Great!
Thanks for coming and chatting last night at the Tech Shop!
I will need to test what's in the box as soon as I can.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2017-04-30 by moz4r