MyRobotLab/InMoov #29: right arm not showing the tabs in gui

hairygael, 2017-02-02:
Hello Anthony,
I tried to launch in a few differnet way to see if I could get it work, but there seems to be an issue with the right arm not starting even though it is activated in my general config and skeleton.
Although the head is not activated it does show the tabs for rothead, neck, jaw, eyes...

Another odd issue is that it always tells me there is a problem with my arduino although it is functional and connected with the correct MRLcomm. Tabs are with servo are worky so I wonder what it detects to be wrong.
I modified the startup sound, I hadn't uploaded the correct one... my bad.
I modified the order of servo for the arm in skeleton according to pin order. I should do the same for the head.

moz4r, 2017-02-02:
Hi!
your rightarm is disabled because mrl cannot connect the com port ( myrightport )
There is a test about if com port is connected or not in mrl
To check if there is a software problem I will delete all, and do a fresh test with last mrl. I tell you soon

hairygael, 2017-02-02:
Thanks,
In parallele, I had tested running just MRL.jar without the batch file and
the Arduino is correctly connected and I can move the servos of the arm
with no problem, so I'm guessing something is going wonky during the
checkup test.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-02-02:
Tested from scratch at home, worky . mmmm so strange ! time to investigate !
LeftSide worky ?
This can happen if myrightport inside inmoov.config is bad 
Can you send a no worky ?

hairygael, 2017-02-02:
Sending a no worky now.
By the way I'm doing the tests with version 1930...

I changed a sentence in the startingLang (fr), to avoid confusion between
servo and cerveau and to let user that it takes a long time to load the
brain.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-02-02:
to be sure COM29 is your port number ?

hairygael, 2017-02-02:
yes it s an extra ADK mega that I use on MinInMoov

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-02-02:
Is this arduino worky from python with an other minimal script ? 
that is very strange . congrats for this problem :)

hairygael, 2017-02-03:
Yes it s worky if I don t run the batch file. you do not reproduce the
issue?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-02-03:
no can't reproduce it ! my Arduino always connect from gui or from python

In this file I have deleted "isconnected" function in the arduino service, to test ( so arduino will always say connected if it is connected or not )

[3_Arduino.zip](https://github.com/MyRobotLab/inmoov/files/749289/3_Arduino.zip)

hairygael, 2017-02-03:
Okay i will test that and let you know

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2017-02-03:
Okay I tested.
The arm servo tabs are now appearing thanks to removing the checkup of
arduino. It works a bit when using the slider and then it stops to work and
I can't detach or attach servo anymore.
Very odd behavior. I have sent another no worky.
I tried also with another pc (tablet), but Arduino IDE 1.8.1 can't upload
the MRLComm on any of my Arduino, Mega, Uno, Nano...
What is that for a mess.
Arduino IDE 1.7.11 works.
But I get the same odd behavior with MRL when launched with the batch file.
If I run MRL without the batch, I can make each servo work properly through
the sliders.

I will stop for now, I have spent too much time today on this and I
seriously need to work now.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

2017-02-03 10:18 GMT+01:00 gael langevin <hairygael@gmail.com>:

moz4r, 2017-02-03:
ok Gael we will find origin of that Strange behavior for sure!
It will be great if next week you bring your arduino and computer, so we can test it in live together. I will bring some hardware too. 
Waiting that day  can you send a zip of your inmoovscript folder please ? I will do more test and try to reproduce the vicious bug, maybe I need to find a mega ADK version too. 
bonne soirée

closed 2017-04-24 by moz4r