MyRobotLab/InMoov #31: Neck auto detached after talking

Doki76, 2017-03-11:
first this is a great script... much easyer to understand then others..
And you can activate one part after another...

i got a problem with speaking and moving the head...
After i use the io1.mouth Service to let him speak it disables the neck (probably more) and my head is falling down... 
more weight at the front of the head and no power in the motor lets him drop the head...
I saw some comments about autoAttach and autoDetach...both are False...but it seems to still autoDetach.

This also happens when i start MoveHeadTimer. This is also stoped after the robot did speak and moved the mouth.

Doki76, 2017-03-11:
I think i found the issue in  MoveHeadRandomize.py.  i did remove enableAutoDetach on all lines...it seems this is always called independed from the config setting...
This should be possible to disable per motor...as my neck motor needs power to stay in position...

moz4r, 2017-03-11:
Hi Doki ! Thank for your report. 
Do you want thing like this ?

[SERVO_AUTO_DETACH]
jaw=True
eyeX=True
eyeY=True
neck=False
rothead=False

Doki76, 2017-03-13:
yes that would be great...
so you can set it dependent if a motor keeps position also when he is not powered

moz4r, 2017-04-30:
sorry about late answer ! autodetach pushed

https://github.com/MyRobotLab/inmoov/blob/develop/Inmoov/inmoovSkeleton/head.config.default#L100

closed 2017-04-30 by moz4r