MyRobotLab/InMoov #32: Disabling different parts in one single config file

Doki76, 2017-03-13:
This is not a technical issue but a proposal of a small redesign of the config files.
Actually i like it that everything is split up in different config files so you can ignore the files that you didn't activate yet. It reduces complexity from a long everything in one file configs...it helps to get used to the setup.

Still i think would be beneficial to have a central config file that contains all switches that turn something on and off... often i just want to test one part or one function and play around with the setting. You would need to go to each other config file and disable it there to turn other parts off.
If you have a central turn on /turn off file you would only need to go there...

This central file could also be a useful guide for starters that provide information on where to fine the detailed config for a particular switch. As config files are distributed in the hole tree i sometimes need to search multiple folders to fine the config i was looking for.

moz4r, 2017-03-14:
Any proposal are welcome thank for it
I agree with you, at the beginning we have hesitated between centralized or not .
A compromise had to be found between light fingerstarter and full inmoov
We are working on a gui to drive this thing from inmoov website. with links to mrl.
It will be very easy to configure and update soon. Maybe config files will be redesigned at this moment.

moz4r, 2017-04-24:
gui wip

closed 2017-04-24 by moz4r