MyRobotLab/InMoov #41: Issue regarding setkey from Rss into NaturalReaderSpeech.

hairygael, 2017-04-03:
Hello Anthony, the develop branch throws an error:
Traceback (most recent call last):  File string, line 44, in module  File InmoovScript/system/InitCheckup.py, line 68, in module    execfile(RuningFolder+services/+filename.encode(utf8))  File C:/mrl/inmoov-develop/InmoovScript/services/5_Mouth.py, line 196, in module    setRobotLanguage()  File C:/mrl/inmoov-develop/InmoovScript/services/5_Mouth.py, line 148, in setRobotLanguage    i01.mouth.setKey(VoiceRssApi)AttributeError: org.myrobotlab.service.NaturalReaderSpeech object has no attribute setKey.

I strongly suggest we keep NaturalReaderSpeech as a third option, because I really don't like having my InMoov speaking with a female voice, and MaryTTS is not yet an option, much too robotic voices still.
:)

moz4r, 2017-04-03:
Hi ! be carefull about naturalreader it is now dead like acapela.  Worky only for cached mp3
Big problem for longtime, we need a reel tts service . Kevin spoke about "amazon poly" it seem a good option . 
I understand I01 & I02 cannot change sex :)
I have pushed a fix ( strange about this error, setkey was conditioned to voicerss only ) . better now ?

hairygael, 2017-04-03:
I will test now.
The thing is that I have modified a lot of various things to make it worky with my InMoov and when downloading a fix, I need to keep all my modifications apart...
For exemple in sleepMode I have:
if isHeadActivated:
		head.neck.setVelocity(60)
		head.neck.moveTo(10)
	sleep(5)
	i01.detach()
	StopNeopixelAnimation()
	sleep(2)
	PlayNeopixelAnimation("Color Wipe", 10, 12, 12, 50)

hairygael, 2017-04-03:
NaturalReaderSpeech works fine at this moment using the master branch though...

moz4r, 2017-04-03:
you can just replace mouth.py if you dont want update all 
Lucky you are, about naturalreader, unstable about copyright warnings
I think my neck is inverted I will mod sleepmode tonight about the code you post , I01 must be the reference.

hairygael, 2017-04-03:
Throwing an error unfortunately. Related to mouth.py
File "<string>", line 44, in <module>
  File "InmoovScript/system/InitCheckup.py", line 68, in <module>
    execfile(RuningFolder+'services/'+filename.encode('utf8'))
  File "C:/mrl/inmoov-develop/InmoovScript/services/5_Mouth.py", line 33, in <module>
    python.subscribe(mouth.getName(),"publishStartSpeaking")
AttributeError: 'NoneType' object has no attribute 'getName'

moz4r, 2017-04-03:
your main config file is ok ?

[TTS]
MyvoiceTTS=NaturalReaderSpeech
MyLanguage=en
MyvoiceType=Ryan

hairygael, 2017-04-03:
It's a clean install. But the main config only shows this:
[TTS]
MyvoiceTTS=VoiceRss
; you can use MarySpeech or VoiceRss
VoiceRssApi=XXX

MyvoiceType=cmu-slt-hsmm
; Voices are automatic donwloaded
; MaryTTS voices - take HSMM ones - http://myrobotlab.org/content/marytts-multi-language-support
; voiceRss use language to change the voice

Nothing about NaturalReaderSpeech in the develop branch.

moz4r, 2017-04-03:
yes NaturalReaderSpeech removed only from comments. But service still here if you want to use it. Do you want add it again into comments for others ?

hairygael, 2017-04-03:
Well lets keep it off for now. As long as I know it's worky, it's okay.
We can close this issue. It's working now.

closed 2017-04-03 by hairygael