MyRobotLab/InMoov #43: buffering overun onRX

hairygael, 2017-04-03:
Sometimes MRL is over buffering, making impossible to go on using MRL or almost. Any idea how this can be controlled?
I must say that this issue is happening when using the master-inmoov branch.

moz4r, 2017-04-03:
this vicious over buffering bug made me crazy ! Stable now from mrl 2000. 2 things :
1/ the problem ( I hope it's yours too ) myrobotlab.org/content/buffer-overrun-cause-script-stuck
2/ master branch is bad, very bad. and it is freezed until someone review it. I cant push modifications. can you review the pending commit ?

hairygael, 2017-04-03:
Master branch can only be pushed by Greg.
Concerning the buffering, it must be because I'm using version 1.0.1970 included in Master branch. And yes, when a script has an issue, it overuns the buffer. Clearing the red warning line sometimes help, but I better get a more recent version.
Thanks for the info!!

hairygael, 2017-04-03:
Ok I reviewed it and approved your commits.
What is inmoovfr tab?

moz4r, 2017-04-03:
thank a lot ! its a good idea to have a stable thing and develop one. The "stable one" was not.
we just need now to make a real good stable version... but people using the master branch will find better things now.
inmoovfr is the optional connector to sebastien gui. I will hide the tab if not activated ( if you dont have key, it is not activated )

hairygael, 2017-04-03:
Ah ok I understand!
Maybe if you call the tab inmoovGui, it will make more sense.

I just tested develop branch with version 2005, and the overun buffering
seems to be gone.

I have the gesture "relax" which is causing a error due to:
if (i01.eyesTracking.getOpenCV().capturing):

It was working before, but I guess something is changed. I can modify the
py and remove that part, but it was handy to avoid the head moving random
when the tracking was on. Maybe you have another plan...?

Question is body random ready to work? I see it on the inmoovGui page, but
I don't have the timer started on my swingGui.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-04-03:
youpi vicious buffer is gone
inmoovGui > ok better name
tracking was the last thing I was testing and not worky good ( using inmoov service commands ).  Chrisitian ,Kevin and others will help us about this I hope ( I understand there is a problem about servo mapping + tracking )
Not enough time from today to do more review. 
You can manual re enable tracking service :
https://github.com/MyRobotLab/inmoov/blob/develop/InmoovScript/inmoovSkeleton/head.py#L165
https://github.com/MyRobotLab/inmoov/blob/develop/InmoovScript/inmoovGestures/minimalHead.py#L36
test command here 
https://github.com/MyRobotLab/inmoov/blob/develop/InmoovScript/inmoovVocal/ear.addCommand/minimal_english.py#L42

good idea to use capturing opencv event to stop random movement using
global RobotCanMoveHeadWhileSpeaking
RobotCanMoveHeadWhileSpeaking=0

body random > todo big list arrrg

moz4r, 2017-04-03:
I just puhed to java land a previous fix that was deleted about head tracking bad pin numbers .
version 2006 worky now wih : i01.startHeadTracking(MyLeftPort,12,13) 
When I say worky it just launch without errors ( previously hand servo was mapped to the head ) but it need to be tested about map and pid errors. Happy to see some friends commit again inside the repo ! see you. really time to faire les valises

hairygael, 2017-04-03:
Great!
Bon voyage!!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-04-04:
develop ---into---> master merge is usually done on release ..
I'd be fine deploying another release - if everyone agrees it would be a good time.

moz4r, 2017-04-14:
Hi grog ! You speak about a new release of mrl ?

hairygael, 2017-04-16:
A new official release would be great!
The last version I tried was 1.0.2006, it was working pretty well, but I
finally broke all the worky stuff, by trying a 64 bit version of java,
hoping it would solve a RX/TX issue I had with NeoPixel.
Lately, I have been so busy, working (work, work) that I hardly find a
moment to run tests with the full InMoov.

Sebastien and Anthony, got the configuration page on the site to simplify
the initial process for the beginners. It's really time to launch something
that is accessible for the beginners.
I don't know if you follow the forum, but there is more and more people
turning over to EZ robot. It's so easier for beginners that I hardly can
debate or argue on that. Unfortunately EZ is not OpenSource and it
frustrate me to see those poor lost souls turning their back to MyRobotLab.
:)

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-04-16:
We can release, I would suggest thorough testing of InMoov, in addition I would really like to bundle the "base" worky InMoov script with myrobotlab.jar .  This works well to keep them in sync.  If all parts are put together and tested as one,  they will **work as one** .

To make it 'easy' a tutorial should be done showing how you startup and use the scripts onboard.  Someone with an InMoov needs to go through a tuto.  The tuto will not work perfectly, and will require help. We can provide the help, but the making of the tuto will drive the focus to make a robust release.

kwatters, 2017-04-16:
If we're going to do a release, I recommend we  have a set of "official" minimal inmoov scripts.  ones that will basically power up everything and allow you to move servos.. and probably also tracking.  Then it's a matter of making the tutorial video for getting it going.. and we could do the release at the same time as the video.  I'd prefer not to rush towards a release if we haven't tested and provided the basic examples.

supertick, 2017-04-16:
Agreed, and these set of "official" scripts must be packaged in mrl in similar fashion as the Arduino ones.  e.g. Start an InMoov service up and a Python tab, comes up with InMoovBasics.py (which is guaranteed to work)

supertick, 2017-04-16:
I think by default the scripts could drive virtual inmoov ...
"As a user, I would like to press a button - which runs a script - which starts virtual InMoov who greets and explains how to run and modify the demo - and how to simply switch to drive a 'real' InMoov"

hairygael, 2017-04-16:
Sure I agree about the minimal scripts. Although as you may have followed
the InMoovOS <https://github.com/MyRobotLab/inmoov>, or InMoov Repo it has
been differently factored, there is only one minimal script:
https://github.com/MyRobotLab/inmoov/blob/master/InmoovScript/Inmoov.py

Because the main configScript
<https://github.com/MyRobotLab/inmoov/blob/master/InmoovScript/Inmoov.config.default>
sets what part you use on InMoov, it doesn't set a minimal script per body
part.
For the other minimal scripts it's here:
https://github.com/MyRobotLab/inmoov/tree/master/InmoovScript/inmoovGestures

Kevin did you try setting the InMoovOS
<https://github.com/MyRobotLab/inmoov> on your robot to see how it works?
I do think we should have some minimal scripts available for each part, but
somehow, all the various possibilities to launch InMoov parts confuse
really the beginners.

I haven't been able to use the virtual inmoov yet, I thought it was due to
my java version, I get a memory java error. But for sure, having a greet
from virtual InMoov explaining demo stuff would be great.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-04-16:
All cooked guys, diner is ready ( some bugs to solve before : input pol for pir, tts, tracking ? )

- vinmoov : At this time a minimal "fingerstarter" is set as default as gael ask as the beginning. but vinmoov can be set by default good idea. Just set ScriptType=Virtual as default https://github.com/MyRobotLab/inmoov/blob/master/InmoovScript/Inmoov.config.default#L4

fingerstarter tuto ( will evolve because of jar integration ) : https://github.com/MyRobotLab/inmoov/wiki/HOWTO---USING-SCRIPT--FINGERSTARTER

minimal tuto : https://github.com/MyRobotLab/inmoov/wiki/HOWTO-USING-SCRIPT-INMOOV-MINIMAL

full inmoov with full gestures ( just activate chatbot,full skeleton and ScriptType=Full )

Optional magical Gui ( about config files headache ) - This will help people to easy feed all config files and configure inmoov ( services, timers, skeleton ... )

kwatters, 2017-04-16:
I'll see about getting a unit test to start up the InmoovScript (hopefully with some virtual arduinos instead of real ones.)  should then be able to have the test exercise all of the gestures .  I'd like to rename some of the directories .. for example.. we should just call it  "InMoov"  not "InmoovScript" ..  but we can leave that until later.

The plan will be to create a InmoovScript.zip and store that in the repo, so when people install the inmoov service, the default scripts are pulled down and installed.

This will be the same script that will run (in virtual mode) as part of the unit test.  I also want to re-enable the jacoco reports and javadoc generation and most importantly... FAIL THE BUILD if the tests don't pass.

supertick, 2017-04-16:
I think it would be worthwhile to focus on integration tests vs unit tests.
Most of the users of MRL use Python and are using it as a tool -not developing it directly.
So, they are interested in what it can provide them, not how it works internally.

We have "service" scripts which should be showing the basic uses of the services in the simplest forms.

Testing these would be optimum.

kwatters, 2017-04-16:
yeah, this inmoov script test will be an integration test. (just doing it with junit.)

moz4r, 2017-04-17:
About vinmoov as default at startup, maybe we can see further on :
- robot : presentation + hello ... ... what is my name
- human : inmoov
- r : I can configure myself, do you want ?
- h : yes/no
- r : I detect 1/2 arduino , what do you want to do with me ?
- h : fingerstarer / full inmoov / modulable
- r: do you want activate pir sensor ?
... ... ...
gui or ini edit will be used to tweak values

that implies chatbot is activated by default too ( vinmoov can disable it after configuration )

moz4r, 2017-04-24:
close initial question

closed 2017-04-24 by moz4r