MyRobotLab/InMoov #44: inmoov tracking service

moz4r, 2017-04-03:
Seem not worky and need refactor about servo mapping
 
i01.headTracking.faceDetect()
i01.eyesTracking.faceDetect()

hairygael, 2017-04-05:
I just tested, it seems to work for me with develop branch. Although the
eyes are not very responsive.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-05-06:
Tracking service must work as a standalone - and it must work with it's service python script
https://github.com/MyRobotLab/pyrobotlab/blob/develop/service/Tracking.py 

It looks like this was changed some by calamity - but the script on the develop branch is not compatible ...

supertick, 2017-05-06:
This service is used often, inside & outside of InMoov .. I think there should be some group discussion regarding how to "fix" and stabilize it.

The topics would include :

- interfaces to support .. e.g. trackFaces, trackPoint, track ???
- initializing .. currently connect() doesnt start the Arduino .. which makes things fail outside InMoov
- current dependencies from InMoov
- multiple axis support ?  - it looks like this was "kind-of" attempted ?

hairygael, 2017-05-09:
Yes I think the tracking service should work on it's own before to be used
within InMoov.
Tracking used to work with:
-trackFaces (was working 107)
-trackPoint (was working 107)
-face recognition (recent versions)

The axis were:
-X and Y eyes (was working 107)
-X and Y head (was working 107)
-Rollneck would be a great addon :)

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Calamity01, 2017-05-09:
The tracking service basically work now, but there is problem with the
initialisation. Look like startPeer use to initialize the servo, arduino
and openCV. required that they are start before and with a very specific
names.

I have made some workaround recently to test the rest of the service and
it's looking good
http://myrobotlab.org/content/tracking-service

supertick, 2017-08-22:
Servo should be cleaned up before Tracking is fixed  
http://myrobotlab.org/content/servo-attention#comment-9226

supertick, 2017-08-27:
Tracking Worky Please Test
I tested on simple pan / tilt kit - worky as simple Tracking service.
tracking = Runtime.start("tracking","Tracking")
tracking.connect("COM5", 6, 9, 1)
tracking.startLKTracking()

Worked for both LK & FaceDetect
[image]
[image]

Please Test ! - If worky for others I'd like to close !

moz4r, 2017-09-08:
robot does not want to look away. I feel spied. I am scared

supertick, 2017-09-09:
Be Scared !

supertick, 2017-09-09:
Have we reached consensus .. Worky or still needs Work ?

hairygael, 2017-09-11:
I will try to test today,
Too worky, worky lately...
Plus I had trouble with my Desktop PC.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

CheekyMonkey, 2017-09-11:
Tested (build 2489) on the Raspberry Pi, but my observations so far are that there is still some underlying PID issue to troubleshoot, particularly in the "y" direction. 

It may be some PID setting local to my setup, but would like to see what others also observe

video [link](https://www.youtube.com/watch?v=kWm8KQ3TabU)

CheekyMonkey, 2017-09-14:
Following further investigation, it appears that the tracking.connect() method may be setting any previously set servo inversion back to default

This then throws out tracking movements for the given inverted servo

A workaround for this behaviour is to, for any servo that requires inverting, call the servo invert method **after** the tracker service connect method
 
Video below showing tracking working in both x and y axis for reference

[Video](https://www.youtube.com/watch?v=rOcIy4UOEFU)

CheekyMonkey, 2017-09-14:
Looking closer at this, I observe that on exiting the following servo method the servo.isInverted() status is reset to False

y.setMinMax(0,180)

So it appears that this is not as a result of the tracking service, but a possible issue linked to  servo.setMinMax

hairygael, 2017-10-05:
Okay tracking head is working nice since version MRL 1.0.2563 for me. It works also good with the inmoov-develop.
Eyes tracking is currently not activated because it still needs some fixing, but I guess for the Manticore version, that's far good enough.

closed 2017-11-16 by moz4r