MyRobotLab/InMoov #45: kinect

moz4r, 2017-04-03:
TODO > add kinect service to script

moz4r, 2017-04-23:
hey from the beginning of the project I had never plugged it, seem worky great I will test gestures based on it, an other story

moz4r, 2017-04-23:
please what is it ?    
i01.leftArm.shoulder.map(0,180,**-25**,105)
i01.rightArm.shoulder.map(0,180,**-30**,100)
i01.copyGesture(True)

moz4r, 2017-04-23:
kinect openni core worky &  pushed into python land
test human skeleton with 360

hairygael, 2017-04-27:
Great it's been so long that I didn't use it...
I wonder if the Kinect2 will work. The tablet being under windows8.1, it's
not compatible. I would need to test InMoov on my Desktop.
These mappings were for InMoov01, for some reason I had to do that in order
to get it working without breaking the shoulder. The kinect had
difficulties to see my arms when I was pointing my arms to the kinect. But
it's been almost 4 years ago and I forgot now.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-01:
i dont have new Kinect2 to test. someone here ?

hairygael, 2017-05-02:
I do have the Kinect V2 and the V1.
I have never been able to make the V2 work because it needs USB3
+Windows10.
I recently purchased a new PC which has all the required, so if i find a
good free time I will have a look.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

juergmaier, 2017-05-04:
Running openNI in mrl and afterwards starting Kinect Studio allows the studio to connect to the mrl kinect service.
With Kinect Studio I get a colored depht view and I can switch the view to "left" and "top". Top looks to be a good start for finding a path around objects?
In current mrl I only get a gray-coded depht-view?

Calamity01, 2017-05-12:
the left and top view are done with processing the kinect data. I don't think it have been implement by someone in MRL. 

But it's possible to do it. I have done a 3d representation of the depth map to use with jme (along with vinMoov). So if you need such kind of data, they are available in the IntegratedMovement service

hairygael, 2017-06-16:
Tested kinectV1 on the Lenovo tablet under Windows8.1 is WORKY!!
I had to remove the mappings in the two commands:
```
i01.leftArm.shoulder.map(0,180,-25,105)
i01.rightArm.shoulder.map(0,180,-30,100)
i01.copyGesture(True)
```
But as I suspected there is some kind of inversion with shoulder, which was already the same issue 4 years ago. The arms at the shoulder joints go the opposite direction than my own arms.
I wonder how to solve that. 
Dominique had added something which Anthony has removed from the OpenNi.py and I wonder if it could be a solution if added into the gesture "startKinect()"

```
if data:
  skeleton = data.skeleton
  leftShoulder = round(skeleton.leftShoulder.getAngleYZ())
  rightShoulder = round(skeleton.rightShoulder.getAngleYZ())
try:
  leftShoulder = int(leftShoulder) - 50
  rightShoulder = int(rightShoulder) - 50
```

moz4r, 2017-06-16:
OpenNi wip , almost worky :) goal is to fix inmoov service and migrate what was inside OpenNi.py . maybe tonight

moz4r, 2017-06-16:
ok seem more solid

what changed : 

Now we have 2 vocal commands about kinect
- START KINECT > start only openni capture
- TRACKING SKELETON > start openni capture + track skeleton

about config :
```
opennishouldersoffset = -50
opennileftshoulderinverted = True
opennirightshoulderinverted = True
```

closed 2017-07-06 by moz4r