MyRobotLab/InMoov #49: pir sensor activation causing mrl freeze

hairygael, 2017-04-05:
On the develop branch if activating the PIR as True, when the SleepTimer tab is launched at startup, MRL window is frozen. Chatbot still respond through Chrome.
Tested with Master branch (before recent merge were done) and I do not reproduce the issue.

hairygael, 2017-04-18:
Tested with latest develop branch, this bug is still alive. I can't send a noworky because of frozen window.
Hopefully you can reproduce this error.

moz4r, 2017-04-18:
reproduced ( but not freeze at home ) something changed about pin adresses. captain informed

moz4r, 2017-04-30:
you said pir worky for you now, I still have issue here

moz4r, 2017-05-06:
just tested still have issue ( worky for gael he is using mega adk , I use mega 2560 )
    >> Arduino.enablePin(23,1)
        at org.myrobotlab.service.Arduino.getMrlPinType(Arduino.java:767)
        at org.myrobotlab.service.Arduino.enablePin(Arduino.java:702)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
it seem this cause the issue : 
if (pin.isAnalog()) 

an idea ?

moz4r, 2017-05-07:
ok, I saw the light but it is far !
I have understand we cannot use anymore arduino autodetection, so we need to use setBoard("atmega2560") inside every script ?
setBoard("atmega2560") is better but not perfect. I have very random issues and some headaches.
Sometime enablepin() worky, sometime not.

This script seem no worky everytime for me ( so that is a good thing ) : 

https://github.com/MyRobotLab/pyrobotlab/blob/master/home/moz4r/Test/enablepin_noworky

log:

01:31:36.331 [python.interpreter.3] ERROR c.m.f.Service [Service.java:2244] python error PyException - null Traceback (most recent call last):
  File "<string>", line 11, in <module>
	at org.myrobotlab.service.Arduino.getMrlPinType(Arduino.java:767)

	at org.myrobotlab.service.Arduino.enablePin(Arduino.java:702)

	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)

	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)

	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)

	at java.lang.reflect.Method.invoke(Unknown Source)

java.lang.NullPointerException: java.lang.NullPointerException

	at org.python.core.Py.JavaError(Py.java:546)
	at org.python.core.Py.JavaError(Py.java:537)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:188)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:204)
	at org.python.core.PyObject.__call__(PyObject.java:496)
	at org.python.core.PyObject.__call__(PyObject.java:500)
	at org.python.core.PyMethod.__call__(PyMethod.java:156)
	at org.python.pycode._pyx3.f$0(<string>:11)
	at org.python.pycode._pyx3.call_function(<string>)
	at org.python.core.PyTableCode.call(PyTableCode.java:167)
	at org.python.core.PyCode.call(PyCode.java:18)
	at org.python.core.Py.runCode(Py.java:1386)
	at org.python.core.Py.exec(Py.java:1430)
	at org.python.util.PythonInterpreter.exec(PythonInterpreter.java:267)
	at org.myrobotlab.service.Python$PIThread.run(Python.java:160)
Caused by: java.lang.NullPointerException
	at org.myrobotlab.service.Arduino.getMrlPinType(Arduino.java:767)
	at org.myrobotlab.service.Arduino.enablePin(Arduino.java:702)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.python.core.PyReflectedFunction.__call__(PyReflectedFunction.java:186)
	... 12 more

Calamity01, 2017-05-08:
> File "", line 11, in

i01.setRightHand() will set the right arduino to "UNO" if you don't specify
the type. So that override the setBoardType you did.

use i01.setRightHand("COM29","mega") or anything that contain the string
"mega" to avoid changing the type of the arduino

moz4r, 2017-05-08:
arduino configuration pushed into specific arduino.config ( because main config was too big ) any comments on this ?

ohhh pir worky great !

moz4r, 2017-05-29:
buuug not killed :) not related to inmoov service.
If we start mrl for first time, with no json inside .myrobotlab :

```
right = Runtime.create("right", "Arduino")
right.setBoardMega()
right = Runtime.start("right", "Arduino")
right.connect("COM29")
right.enablePin(23,1)
```

this crash. if we reopen and launch the script again it worky and pinlist is ok

moz4r, 2017-05-29:
fixed ( definitely I hope )

closed 2017-05-29 by moz4r