MyRobotLab/InMoov #51: Rename inmoovScript to inmoov

moz4r, 2017-04-18:


moz4r, 2017-04-19:
Hi @kwatters ! about your question can we rename inmooscript to Inmoov without problem ? because there is already an inmoov folder at root ( used by jmonkey )

Calamity01, 2017-04-19:
the inmoov folder hold the 3dmodel for the VinMoov. So it's all relate the
the InMoov stuff. It can probably be shared or if you think it's better, I
can move the vinMoov stuff elsewhere

moz4r, 2017-04-19:
mix inmoov stuff is good idea ! I don't know how the folder from inmoov repo will be merged into the jar

moz4r, 2017-04-24:
ok inmoovScript renamed to Inmoov and merge existing Inmoov folder while install

moz4r, 2017-04-30:
kwatters: gestures directory should be renamed from "inmoovGestures" to just "gestures" ...
kwatters: (not sure why there's a subdirectory for COMPLETE_GESTURES ...
kwatters: all gestures should be in the same directory.

moz4r, 2017-04-30:
InMoov as folder name

moz4r, 2017-04-30:
we need to put somewhere minimal gestures, it is usefull for people using minimal inmoov. Minimal inmoov is no chatbot and few skeleton parts. I propose a new folder Minimal , inside we put basical globalised ear commands + minimal gesture.

moz4r, 2017-04-30:
guys, whole project renamed. I hope I didnt broke anything. seem ok but not tested all. pushed virtual inmoov by default ( with right hand ) instead of fingerstarter inside configuration. Everyone ok ?

closed 2017-05-09 by moz4r