MyRobotLab/InMoov #52: TrackHumans et TrackPoints comeback to origin

moz4r, 2017-04-18:


hairygael, 2017-04-18:
For now these settings are giving good results although the eyes could be even faster.
def trackHumans():
  global RobotCanMoveHeadWhileSpeaking
  RobotCanMoveHeadWhileSpeaking=0
  i01.headTracking.faceDetect()
  i01.eyesTracking.faceDetect()
  i01.setHeadVelocity(80, -1)
  i01.setEyeVelocity(-1, -1)

What about you?

moz4r, 2017-04-18:
setEyeVelocity not reconized I changed to i01.setHeadVelocity(80, -1, -1, -1, -1)
Lucky guy !! bad results for me ( using 2042 ) . I need to do more test. I will open a post after

01:00:10.687 [i01.eyesTracking] ERROR c.m.f.Service [Service.java:1514] InvocationTargetException
java.lang.reflect.InvocationTargetException: null
	at sun.reflect.GeneratedMethodAccessor51.invoke(Unknown Source) ~[na:na]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source) ~[na:1.8.0_121]
	at java.lang.reflect.Method.invoke(Unknown Source) ~[na:1.8.0_121]
	at org.myrobotlab.framework.Service.invokeOn(Service.java:1445) [myrobotlab.jar:na]
	at org.myrobotlab.framework.Service.invoke(Service.java:1383) [myrobotlab.jar:na]
	at org.myrobotlab.framework.Service.run(Service.java:1794) [myrobotlab.jar:na]
	at java.lang.Thread.run(Unknown Source) [na:1.8.0_121]
Caused by: java.lang.NullPointerException: null
	at org.myrobotlab.service.data.Point2Df.get(Point2Df.java:44) ~[myrobotlab.jar:na]
	at org.myrobotlab.service.Tracking.updateTrackingPoint(Tracking.java:573) ~[myrobotlab.jar:na]
	at org.myrobotlab.service.Tracking.setOpenCVData(Tracking.java:385) ~[myrobotlab.jar:na]

hairygael, 2017-04-18:
oops sorry,
this is the correct one:
def trackHumans():
  global RobotCanMoveHeadWhileSpeaking
  RobotCanMoveHeadWhileSpeaking=0
  i01.headTracking.faceDetect()
  i01.eyesTracking.faceDetect()
  i01.setHeadVelocity(80, -1)
  sleep(1)
  fullspeed()

moz4r, 2017-04-27:
TrackHumans et TrackPoints comeback to origin,  check !

closed 2017-04-27 by moz4r