MyRobotLab/InMoov #54: activate virtual inmoov + full inmoov

moz4r, 2017-04-18:


hairygael, 2017-04-19:
> ok pushed ! Inmoov and his virtual friend can work together
>
> 06a10b0

Hello Anthony,
Great!
I just installed the latest develop branch to check.
VinMoov is activated along with InMoov, but it doesn't actuate when moveing
servo sliders or voice commande gestures.
Do you have it working simulteaneously?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2017-04-19:
Ah sorry I found what I did wrong. Forgot to copy a line in the checkupini.py

moz4r, 2017-04-19:
I push mod at work work , just tested now, result is so cool !!!

hairygael, 2017-04-19:
It's worky for you???
Arrgh

hairygael, 2017-04-19:
Tested VinMoov with the French version and it works partially.
The arms are moving and working, but not the rothead, neckhead, jaw, hands and wrists.

moz4r, 2017-04-19:
ok I will look forward on it, maybe an init delay problem I remember @Calamity01 spoke about something like this, maybe not related

moz4r, 2017-04-30:
is it better now ?

moz4r, 2017-05-01:
set full skeleton activated by default if scriptType=vinmoov

hairygael, 2017-05-02:
My vinmoov is not moving anymore with version 2132.
I will need to investigate.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-10:
Some good news here ?

hairygael, 2017-05-10:
I didn't have the time to test with a more recent version of MRL...

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Calamity01, 2017-05-12:
vinMoov is working fine with my own inMoov script with MRL 2161. So we should look in the inMoov repo script for the bug hunting

hairygael, 2017-05-12:
I tested last night with 2156, and it works with the repo when selecting Virtual.
Though the right wrist is still not turning the correct direction.
I had a look to see what you suggested to modify that in the inmoov.java and also in the jml3.java.
But I do not want to brake something in there instead of fixing it.

Calamity01, 2017-05-12:
add the following python line in the script after vinMoov is started

i01.vinMoovApp.setMinMaxAngles("RWrist",180,0)

that will set your min of the servo to = 180 degree in the vinMoov reference and max of the servo to = 0 degree in the vinMoov reference

I have refer to the method so you can find the proper names of the part

hairygael, 2017-05-12:
Thanks Calamity,
Indeed that would fix for my personal instance, but all the beginners might think they didn't set their mappings correctly.
Should this line be modified:
maps.put("RWrist", new Mapper(0, 180, 0, 180));
to
maps.put("RWrist", new Mapper(0, 180, 180, 0));
https://github.com/MyRobotLab/myrobotlab/blob/develop/src/org/myrobotlab/jme3/InMoov3DApp.java

Calamity01, 2017-05-13:
no, that line is just a linker so I could make your way of mapping servo to work with vinMoov. I eventually want to remove that line

Better work on how the joint is define
a few line before that line, you will see

rotationMask = ...   the value -1 or 1 define the orientation of the rotation (clockwise or counterclockwise)
angle = define the initial position of the part (rotation of the 3d model in his local world)

playing with those two value is better in the long term

moz4r, 2017-05-16:
follow the friend post : https://github.com/MyRobotLab/myrobotlab/issues/71

closed 2017-05-16 by moz4r