MyRobotLab/InMoov #55: Mixed finger servo pin

hairygael, 2017-04-19:
Just discovered a bug which cause a conflict. I'm not sure if it's MRL related or the InMoov repo.
[image]

moz4r, 2017-04-19:
I think I know what is your problem : when we detach  a servo , the combobox reset to the first arduino available, the left. I will try to fix it

moz4r, 2017-04-20:
ok pushed ! but I have problems with last last mrl ( seem due to attach/detach actualy refactor ) 

https://github.com/MyRobotLab/myrobotlab/commit/752c6c43513adf361cd0ccffc83c8fdf8104e420

hairygael, 2017-04-20:
Okay thanks!
I had tried version 1.0.2040 and it was too crazy with errors. Though I used that version with the Master branch, so it might have been causing the issues.
So you recommend not to download the latest MRL yet?

Calamity01, 2017-04-20:
Let me explain the attach/detach refactor.

there is currently two differents action that are named
servo.attach()/detach(). One of them (servo.detach(), servo.attach() and
servo.attach(pin) is named after the arduino servo library
attach()/detach(). Those method are mean to turn on/off the servo power,
and should keep everything relate to the servo usage active in both MRL and
MRLComm.
the method servo.attach(controller, pin, position) and
servo.detach(controller) are mrl methods that allow to plug and initialize
a service to another one or completely remove it's instance.

Those two functions are differents, but their naming bring a lot of
confusion. So we decided to change the name of the method to turn power
on/off and it will become servo.enable()/disable().

I have start doing the change, but look like it raise up problem with the
mixup between the two actions, probably in the inMoov services,  in the
inMoov script and/or the servoGui. I will try to fix that this week-end

hairygael, 2017-04-20:
Oh Boy! Thanks for the explanation Calamity!
We are not ready to have a worky version.
:)
Many gestures.py including AIML of InMoov have some attach() and detach()

At which version of MRL did you guys start to refactor attach() detach(),
maybe we should propose that version for the builders that need a current
worky version? I'm using the 1.0.2006 and it's pretty okay beside the fact
servo controllers get mixed up if we detach servos.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Calamity01, 2017-04-20:
The plan is to make attach() detach() as @Deprecated but worky for a while
to keep backyard compatibility. So you don't have to worry about it.  But
if the gesture script have something like attach(arduino, pin) or
detach(arduino) you are probably not using the right method (unless you are
initializing a servo not support by the inMoov services, like the tilt head
servo) But that should not be in the gesture script.

I think more the mixup are in the InMoov services and the goal is to end
those mixup that still happen even since we try to remove them for a long
time

moz4r, 2017-04-21:
arrrg no no dude !
Ok do what you want ... what you make is so great :)  

for information there is attach(arduino, pin) everywhere in the script !!! 

[image]

why : to give possibilitie to users to overide hardcoded inmoov pins. And like you say to support extra servo mods. 
ok, I will remove it and tweak a thing inside inmoov service later
if only standardized inmoov worky , it is a good start. 
So time to feed inmoov service.

@hairygael I have pushed eyelids and set pins default to 29/30 ( I set 2 pins but usually only one used )
about rollneck i will cook it too soon ( hacked servo ). 
what do you want about all those extra pins ?

Calamity01, 2017-04-21:
maybe I badly explain

using servo.attach(arduino, pin) is fine for like what you said, overide
hardcoded pins and support extra servo.   After that setup. if you want to
temporary turn off the power to the servo you use servo.detach()... soon to
be servo.disable(). and repower the servo with servo.attach()... soon to be
servo.enable()

if you ask the question, it's because it's not clear that
servo.attach()/detach() are different thing than servo.attach(arduino,
pin)/detach(arduino) and that's why we want to rename
servo.attach()/detach(), to be clear it's two different things

Calamity01, 2017-04-21:
let me give you an concrete example.

You initialize a servo and you connect it to the arduino using
servo.attach(arduino, pin). This will send a message to mrlcomm to create a
device to make a servo working.

after a bit you want to turn the power of the servo off.
you should use servo.detach()/disable().  This will send a message to
mrlcomm to do servo.detach(). With that command, the communication with the
servo is keep, the servo just don't have any power to move.
But you can still set the velocity, you can still ask to moveTo() (the
servo will obviously dont move, but when power get back on, it will move to
that position)

If you use servo.detach(arduino). The servo will turn off, but that will
also remove the device for the servo in mrlcomm. You can communicate with
the servo until you recreate the device with servo.attach(arduino, pin) and
will need to set the velocity and it's position

I hope this will help clarify this things

supertick, 2017-04-21:
Just to give some background on the issue ... it's a **name conflict**  between Arduino's Servo.attach and  MRL's "attach"
http://myrobotlab.org/content/attach-pattern 
Attach/detach is a **very very important** concept to MRL.  You can read in detail from the link about what we are trying to do, but in short "Attach is what one service can do with another"  - Like Blocks !
Regrettably, Arduino library used this name for enabling and disabling the pwm for servos - and much confusion and problem arose....   What happened is the servo was detached from the Controller and that's not what was desired.

So rather than breaking all the other attach interfaces - e.g. attaching ear to mouth, speech synthesis to speech recognizers, device controllers to device controls, platform controllers to joystick controls, and on and on..   We decided to say Servo.enable()  .. Servo.disable()
This seems also fairly consistent with enablePin() & disablePin()  .. and other device enable & disable calls on the MrlComm level..

This has been discussed at great length ..

moz4r, 2017-04-21:
Thank guys about those precious informations. All is clear now

[image]

hairygael, 2017-04-21:
Nice! It makes sense.
So actually basically it will stay the same. For exemple, In my gestures I
will only rename to i01.disable() to just turn off the power for the whole
robot but keeping all servos attached to the controllers.

Question: what happens if I move a servo slider while disable and then I
enable it. Will it move suddenly to reach the position of the slider? Same
question, if I send a voice command (DaVinci) while disabled, what happens
when re-enabled?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Calamity01, 2017-04-21:
The servo that are disable act the same way as enable servo, except that
the horn won't rotate. They will continue to keep track of their position
they should be, velocity, min/max etc. Basically, they become a ghost servo.
When you enable an servo that have been disable, it will move to the
position it was last told to move.

Not sure it's the best behavior, but that's what it currently do.
If we think it's better, it should be easy to reject moveTo() call while in
disable mode so when the servo get enable again, it stay at the same place
that where it was when it got disable

I have not think about it before, but I think the second option make more
sense. What do you think?

moz4r, 2017-04-21:
Don't worry about gestures, modifications about servo values will occur to low level 
maybe it is better to overide the name of i01.detach() too, so we use i01.disable() ? 

About servo , by default in the script there is automatic attachment( enable ) everytime.
So if you move a slider, servo will poweron automatically.
No need vocal command anymore about this ( only to force a detachment ( disable ) )

can you check this, if you agree about the idea of head integration of the rollneck : http://myrobotlab.org/content/inmoov-bom-update-new-skeleton-parts

moz4r, 2017-04-21:
ohh same time we post :)

hairygael, 2017-04-21:
@calamity, not sure the second option is the best...The first is more logic
to me, (maybe because it always worked that way)
@Anthony, answered your post on MRL. Do you think we could add the
autoAttach, autoDetach (yet to become autoEnable, autoDisable) per servo
instead of per body parts?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-04-22:
yes of course
You want want to set some specific servos to autoDisable by default ? hardcoded or inside config ?

hairygael, 2017-04-22:
Hello,
Hardcoded would be the best, because some people will not use the config
and they still could benefit of that option in the service.
What do you guys think?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-04-23:
Tell us witch servo you want disable after a moveTo and it is like it's done

hairygael, 2017-04-27:
If it's not too much work to add the option for each servo in the config
files, it would be nice.
Somehow most servos could be autodisable, but some times it might not be
handy for some tests if it's hardcoded in java without True/False option.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-02:
option added, forgot to comment this

closed 2017-05-02 by moz4r