MyRobotLab/InMoov #60: Enter gestures keyword

moz4r, 2017-04-23:
possible to override it ? to launch gestures without enter the magical keyword ?

hairygael, 2017-04-27:
We can try, I'm worried that with English version chatbot, which is much more complex than the French chatbot, it might end up answering things to the voice commands instead of doing the requested gestures.

By the way, I tested the French chatbot and it doesn't work really well, it uses very often default response instead of using the possible AIML responses. Do you have the same?

moz4r, 2017-04-27:
great ! will think to add an option , but not easy to tweak this part.
about french chatbot totaly agree it is unstable state , there is so much work to do , we are alone to feed it.
If you have some examples to push some updates...
My goal at the begining was to work on it , at the end I will return :)

moz4r, 2017-05-14:
removed magical word , to test

closed 2017-05-14 by moz4r