MyRobotLab/InMoov #61: Speed problems

moz4r, 2017-04-27:
this post to enumerate speed problems.
To know : setspeed is deprecated but still worky.
To convert setspeed to setvelocity here is a tip 

		// 0.1 3
		// 0.2 5
		// 0.3 7
		// 0.4 9
		// 0.5 13
		// 0.6 19
		// 0.7 26
		// 0.8 36
		// 0.9 50
                // 1 -1

Calamity01, 2017-04-27:
to be more exact you should use that formula to convert old speed value to
velocity

velocity = e^(3.25 * speed + 1)

Look wierd, but it's the formula the explain the best the relation between
the two

moz4r, 2017-04-27:
maths !!!

[image]

thank you :)

hairygael, 2017-04-28:
So maths is taking us beyond understanding!!
@Calamity, there is a serious issue with speed gestures in version 1.0.2105. All my gestures are acting really too slow although all setMaxVelocity are at -1. 
Do you know what could be the reason?

Calamity01, 2017-04-28:
I have change nothing about setVelocity/speed/maxVelocity in ages, so I
don't know. I will try to inspect that.

hairygael, 2017-04-29:
Thanks Calamity,
it's very odd. The gestures are slower since a while though, but it seems like the gestures cannot even finish and reach their positions because of delay.

moz4r, 2017-04-29:
Off course we need math, I was just a bad student :)
I have set davinci() like it was before : using setspeed
and disable autodetach inside this gesture just to test.
https://github.com/MyRobotLab/inmoov/commit/a9a71a3ca7bbbc1d0573ee751f850cebb1f97fa7

better ?

I propose a thing about autodetach :  
Using a timer instead of virtual endenstop calculation ( like if using autodetach if velocity = -1 )
So no problem anymore about load

Nice weekend guys

hairygael, 2017-04-29:
Hello moz4r,
Didn't test it yet because all connections are disconnected on my InMoov beside the hand. I'm testing a minimal script with the hand to see what are the main problems I encounter.
But as soon as everything is connected back, I will test da vinci.
Your autoDisable proposition could be a trick to prevent premature disabling of the servo.

Calamity01, 2017-04-29:
if the servo detach before the movement ends, it's because the velocity
setting is too high and your servo can't reach the speed you specify

There is no math in the autodisable function. The only math is about the
computation of servo based on the velocity (after x ms, the position of the
servo is y). When the arduino complete it's moveTo() it send a message
saying that the movement is complete and MRL disable the servo. Of course
if the velocity is set too high, the arduino will think he complete his
movement too early. Unless we can get feedback from the servo, that would
always be a problem. But from the test I did some time ago, it was working
well.

I think that may be more relate to the fact that the servo are moving
slower than expected. You say that many time recently. We should
investigate that before when when fixed, I think the autodisable will work
as it should again without modification

Calamity01, 2017-04-30:
closing this ticket as the problem is not with speed but with autoEnable/Disable.
And I think the "speed" problem is already resolved

closed 2017-04-30 by Calamity01