MyRobotLab/InMoov #65: mapping issue leftArm.omoplate

hairygael, 2017-04-30:
I have a mapping issue regarding leftArm.omoplate. Although it is correctly mapped, same as the right arm, the mapping is not active.

moz4r, 2017-04-30:
tested to increase value with my cobaye, left omoplate dislocate good :)
When you say not active, omoplate physical range still 0>180 ?

those are my values

[SERVO_MINIMUM_MAP_OUTPUT]
omoplate = 10

[SERVO_MAXIMUM_MAP_OUTPUT]
omoplate = 90

[MINIMUM_MAP_INPUT]
omoplate = 0

[MAXIMUM_MAP_INPUT]
omoplate = 180

hairygael, 2017-05-01:
These are my values for the leftArm, I have almost similar values for the rightArm and it works fine.

[SERVO_MINIMUM_MAP_OUTPUT]
omoplate=42

[SERVO_MAXIMUM_MAP_OUTPUT]
omoplate = 80

[MINIMUM_MAP_INPUT]
omoplate = 0

[MAXIMUM_MAP_INPUT]
omoplate = 180

So the armLeft omoplate when doing Da Vinci only moves about half way up and stops. What is odd, is the slider. if I go to 80 which should be the end stop, now I can go more and the arm could dismantle and go much too far.

moz4r, 2017-05-02:
can you check with servo_calibration ,left omoplate range values ?

hairygael, 2017-05-04:
Both arms have the same config values and you can see the difference given in the sliders:
[image]
[image]

moz4r, 2017-05-04:
I hope there is a difference ( or it is black magic ! )
maybe there is a difference between leftarm and rightarm about omoplate inside 
[MINIMUM_MAP_INPUT]
[MAXIMUM_MAP_INPUT]
We can see in your screenshot values are not the same.
I have reproduced too a similar issue :  

[image]

other possible issue : some error in config file, so config not parsed
No subconcious voice error at startup about bad config file ?

can you post your skeleton config ? So I can look at it more further

hairygael, 2017-05-04:
Gee, I found the issue!
Thanks for overlooking! Some update as modified my config 
[MINIMUM_MAP_INPUT]
[MAXIMUM_MAP_INPUT]
The left was set to 0-->180 instead of 10-->80
Looking in a older folder my values were okay, so I don't know what happened.
Anyway its all good now.

moz4r, 2017-05-04:
great ! no more dislocated skeleton it hurt too much. 
I understand you use input to set your limits, so your map is map(10,80,0,180) instead of classical map(0,180,10,80) . but if worky don't touch nothing ! other post other day i m not an expert about angle standardisation

hairygael, 2017-05-04:
My map should be map(10,80,42,80) for leftArm.omoplate. But on the full script I have leftArm.omoplate.map(0,180,42,80)
Me neither, there is things I do not understand...

hairygael, 2017-05-04:
By the way, does the jaw works okay for you?

moz4r, 2017-05-04:
this is my mouthcontrol freshly tested to compare

https://www.youtube.com/edit?o=U&video_id=wBaLPyU2xJQ

closed 2017-05-07 by moz4r