MyRobotLab/InMoov #66: Downloading a new voice

hairygael, 2017-05-01:
Downloading a new voice for MarySpeech hangs after it's finished in the terminal.
The process is actually done and finished, but it looks in the terminal like the user needs to wait because it says PROCESS1/1 and just waits.
It's only a bit confusing and I have no idea if that can be arranged. Idealy would be autoterminate and restart. That would be in Wonderland!

moz4r, 2017-05-01:
subconscious robot voice had logically supplied you to do a restart .
2 things I propose to be in wonderland
- put all supported marytts voices jar inside mrl lib folder so no need to download voices anymore, until marytts next update
- have an idea to autorestart things but want to ask captain @supertick if the agent can autorestart ( exit instance, relaunch mrl+script ) and how to do it

hairygael, 2017-05-02:
It might be heavy to load mrl with all the Marytts voices. Captain will say.
It would also be great to have another (better) voice for default, like the
cmu-bdl-hsmm. Because cmu-slt-hsmm, is almost impossible to understand what
she says.
Hopefully the Agent can do the necessary autorestart.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-10:
cmu-bdl-hsmm added, without download needs

hairygael, 2017-05-10:
Nice!
The InMoov repo needs to merge to master again i think, because Master
doesn't work anymore. (too many changes.)

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2017-05-12 by moz4r