MyRobotLab/InMoov #67: Add german language pack, thank juerg for help

moz4r, 2017-05-01:
in progress

moz4r, 2017-05-01:
@juergmaier congrats your are now official Inmoov german personal teacher :)

hairygael, 2017-05-02:
Great Juerg! That is really cool!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

juergmaier, 2017-05-04:
german "dfki-pavoque-natural-hsmm" is worky for me now - needs the voice jar in the library/jar folder however.
I have also added a "de" chatbot which works for me. It is a slightly modified version of Christian Drossman and so far has no integrated specialized InMoov parts. These would anyway need to got into an additional .aiml file. The german file is split into 2 files because the single file was too large for mrl to load.
What I don't know how to do is where the additional language has to be added (currently en,es,fr) in the config files in order to get distributed with new downloads. 
Anyway - have a problem to see how my personally modified config settings (language just one the them) will get merged with new/modified versions of the file in the repo.

hairygael, 2017-05-04:
If you have some time, it would be nice if you can translate two files
which will be added in the AIML bot files.
These files contain the switch between AIML and Gestures already pre loaded.
And also it sets the initialisation of the robot.
Regarding the currently en,es,fr, I think Anthony will take care of that,
as he knows where to add the "de" config.
https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/chatbot/bots/de/aiml/_inmoovChatbot.aiml
https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/chatbot/bots/de/aiml/_inmoovGestures.aiml

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-17:
personal modified config settings are not merged and safe. 
If you update mrl your config settings and chatbot predicates are not overwriten. because what is pushed is only default config files. if not exist.

moz4r, 2017-08-26:
this can reborn later; no fresh updates

closed 2017-08-26 by moz4r