MyRobotLab/InMoov #71: Shutdown actions

moz4r, 2017-05-09:
Some shutdown actions can be implemented. I'm thinking of one useful thing , is to set whole robot rest() just before shutdown. Why : because next time started, servo will not move at fullspeed to rest(). 

shutdown  actions :
- rest()
- disable servos
- disable neopixel ...

hairygael, 2017-05-09:
This is actually what I do most of the time.
-Keep the head going down though, this shows it's sleeping with rest()
-disable servos
-keep a sleeping neopixel animation. or the choice in the config?
-wake up mode, I find it strange that you have set a random. If we have a
random it should be on what it says but not various actions.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-09:
There is two things here
- Sleeping mode
- Shutdown actions

- Shutdown actions : when you close mrl using the cross , or tell robot to shutdown and exit ( not sure if this keyword implemented, it will )
At this time mrl just close and leave things powerup, the idea is to rest(),disable and poweroff things as poosible we can. head down can be set off course -> but need to set at startup too to avoid initial head landing to the moon.

- Sleeping mode : I personally like random things in this scripted world :) off course when it is not random errors arrrrg 
ok do you want a personal config parameter ?

Calamity01, 2017-05-09:
Shutdown actions can be split in two situations. Controled shutdown and
after crash shutdown.

Controled shutdown can be view as the user turning off the inMoov, can be
treat as a gesture or something similar. It can also be from the user
quitting MRL with the X button, in that case maybe the releaseService()
method of any Service can be use. But I don't know in wich order services
are release on a MRL shutdown

after crash shutdown should be a reaction that happen in Arduino to go in a
safety state when something bad happen (computer or MRL crash, usb cable
disconnected (but arduino still powered etc). So MrlComm should disable
servos, stop motor (like diy servo), stop reading pins, stop or show a
special animation for the neopixel etc. Data in the MRLComm are limited so
it can only be simple things.

closed 2017-05-23 by moz4r