MyRobotLab/InMoov #72: jm3 directory not loaded

hairygael, 2017-05-11:
Using MRL 1.0.1156, the jm3 directory isn't present inside the InMoov directory, is it normal?

moz4r, 2017-05-12:
No :) 
Sometime when I update the repo script ( because what is Inside the jar toooo old ) I delete by mistake the jm3 folder. Maybe you do same mistake ?
tested Inside 2163 jm3 is here when Inmoov service installed

hairygael, 2017-05-12:
I will need to see with the latest build to see if it's in it.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-17:
can we close it ?

hairygael, 2017-05-17:
Just checked with latest build 2173.
Directory is loaded with " install all". I didn't test to see if install
inmoov only loda the folder though.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-17:
great 1 less bug !

closed 2017-05-17 by moz4r