MyRobotLab/InMoov #74: todo : inmoov brain documentation

moz4r, 2017-05-12:
need a tool to create some of shared diagram.
It will be used to document some random reactions of the robot, inmoov service, chatbot functionalities...
Usefull for new developers and us before we get lost inside the code...
Any ideas guys ? necessary ?

moz4r, 2017-05-17:
https://www.draw.io/?client=1&title=Copy%20of%20Untitled%20Diagram.xml#HMyRobotLab%2Finmoov%2Fdevelop%2Ftools%2Fshared_diagrams_documentation%2FInMoovLife.xml

Sample : 

[image]

hairygael, 2017-05-17:
> https://www.draw.io/?client=1&title=Copy%20of%20Untitled%
> 20Diagram.xml#HMyRobotLab%2Finmoov%2Fdevelop%2Ftools%2Fshared_diagrams_
> documentation%2FInMoovLife.xml
>
> Sample :
>
> [image: image]
> <https://cloud.githubusercontent.com/assets/18124594/26159800/d92f7d72-3b1f-11e7-8d98-df3da6be4d81.png>

Nice diagram Anthony!
Great work!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

juergmaier, 2017-06-01:
Hi Anthony
I am still working on the "open the door" scenario and work on the path finding and negotiating to a target location. In your diagram you have a PIR sensor that is specifically designed for detecting humans?
In my case I would like to command my InMoov to "go to the door". I will mark the door with a marker. I would like to use aruco-markers but so far have not been able to find/create a python library from the existing library I found on the net. I am kind of limited to python because all the other stuff I have done so far (kinect, path finding, visualisazion with pygame) I have done in python 2.7.
Do you have any experience with python wraps for c++ libraries?
http://docs.opencv.org/3.1.0/d5/dae/tutorial_aruco_detection.html
Juerg

moz4r, 2017-06-06:
Hi juerg ,  great idea ! look forward to test all of this. I saw grog said we need javacv 1.32
Yes pir specifically designed for human detection , to poweron opencv or wakeup the robot

moz4r, 2017-07-12:
https://github.com/MyRobotLab/inmoov/wiki/BRAIN-DOCUMENTATION

closed 2017-07-12 by moz4r