MyRobotLab/InMoov #75: French _inmoovGestures

hairygael, 2017-05-14:
@Anthony, 
hello,
Why are all the voice commands to enable parts are removed from _inmoovGestures?
Do you plan to put them someplace else?
I personaly use these commands all the time when working on the robot, specialy if I do student sessions. I will be doing one very soon in Brussel and was counting on using French version.
For exemple if you are working only with a arm or a hand (not having a complete inmoov) it's great to be able to voice enable or disable each components.

I'm guessing, it's because you consider the enableAutoEnable as a pertinent use. 
Although it is totally effective when running a complete robot, it might not be handy when working on the robot, therefore voice controlling the connections remain usefull.

moz4r, 2017-05-14:
Hi Gaël ! every inmoov servo use now AutoEnable by default , so voice command to enable it is totaly useless.
AutoEnable worky great everytime ( AutoDisable still experimental )
There is still "disable" vocal commands. So you can disable parts like yesterday. No need to say "enable" anymore because it is fully automatic.
But those vocal commands can be repushed, it was just to cleanup useless things. as u want

hairygael, 2017-05-18:
Hello Anthony,
I understand. I'm thinking also about script usage. It is important to keep compatibility between a full script and the inmoov repo. Somethings might be causing issues if they are removed from the repo if you are running a script to launch InMoov , while not using the configuration inmoov repo.
Otherwise we may keep having issues with people trying to launch the full script using all the components of inmoov repo(e.g: chatbot, gestures...)
Do you see what I mean? It's not always easy to explain stuff by writing.

moz4r, 2017-05-18:
Hi Gael , I understand, it is a necessary logic . 
About "attach" we have 4 options 
- to evolve inmoov service ( 2 or 3 lines ) so every script will use autoattach by default,and no backward compatibility problems.
- repush those vocal commands
- evolve+repush
- ask inmoov what he think about

as you want

hairygael, 2017-05-18:
Ehehe! Ask InMoov.

-evolve+repush (is more appropriate, I really use those commands even if
enableAutoEnable is working nicely, because it allows to connect only a
piece of the robot vocaly. I use it all the time when working on the robot.)

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-23:
pushed

closed 2017-05-23 by moz4r