MyRobotLab/InMoov #76: vinmoov error

moz4r, 2017-05-16:
sometime I have this error at startup
maybe an init delay problem

[image]

Calamity01, 2017-05-16:
> Assigned #76 <https://github.com/MyRobotLab/inmoov/issues/76> to
> @Calamity01 <https://github.com/calamity01>.

There is a kind of guard that put the inMoov service to wait while the jme3
engine start up to 5 sec. When the jme3 finish loading (end of
simpliInitApp()) it wake up the inMoov service for further interraction
with the 3d app. You add things to the 3d app, do it now can take more than
5 sec to initialized the 3d app. If so, changing the line wait(5000) to a
bigger number or just use wait() may help

But the error messages you show look more like the Root Node is been
modified outside the 3d app thread (mean not in simpleIniApp() or
simpleUpdate() method)

But I will need to finish my work work rush before I can look further to
that

hairygael, 2017-06-01:
Hello,
running version of MRL 1.0.2226 with inmoov-develop without changing any configuration.(all default)
VinMoov is launched but doesn't work with gestures or sliders.

On another topic, it seems that VinMoov works only on 64 bit computers, because java 64 bit is required.
Maybe there should be an alert for users with 32 bit computers.

moz4r, 2017-06-26:
There is an alert for 32bit .  https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/services/F_VirtualInmoov.py#L7
The question is VinMoov really not compatible with 32bit ? Someone can confirm ?

hairygael, 2017-06-26:
> There is an alert for 32bit . https://github.com/MyRobotLab/
> inmoov/blob/develop/InMoov/services/F_VirtualInmoov.py#L7
> The question is VinMoov really not compatible with 32bit ? Someone can
> confirm ?

On my 32 bit it is NOT working. Now why is that not compatible, I have no
idea.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-08-26:
works on my 64 bit system

supertick, 2017-08-26:
jm3 uses jni for some of its display - I'm guessing opengl jogl stuffs ...
since JNI is evil but necessary for fast graphics, it takes work to make compatible for every operation (vs pure Java which is already compatible) .. 
I don't have a 32 bit system to confirm unless I try it with a RasPi 1 .. (heh even at this point RasPi 3 is 64 bit :)

hairygael, 2017-08-26:
Thanks Greg for the info.
As for now it worked on all my 64 bit systems under windows. I never tried
on my Raspi though.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-09-14:
Ahoy !  
Please advise if this can be closed ...  It got a little mixed with other topics - so I'll try to summarize

- 32 bit does not support Virtual InMoom - I am not interested in trying to solve this as all 32 bit hardware is dead to me ... for supporting graphics
- if there is an init problem, perhaps Calamity solved it with more of a wait time ... typically we want semaphores to signal when things are properly initialized, but if Calamity is good with it I'm good with it

Yay or Nay ?

hairygael, 2017-09-14:
Hello!
Ok I understand about the 32bit being dead. Too bad some of our lenovo
tablet running InMoov are based on 32bit.

There is still an issue with VinMoov but it's related to the rotation of
servo being inverted on VinMoov which is posted in another error.

We can close this.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-09-14:
32 bit lenovo work great .. for controlling servos, motors, looking at sensors, doing everything ! 
But not for graphics ... next release will have **very powerful connectivity** - so don't throw our 32 bit friends out !  They will be part of the borg again :)

Closing ...

closed 2017-09-14 by supertick