MyRobotLab/InMoov #77: Inmoov control center

moz4r, 2017-05-17:
Sébastien our favorite webmaster had done some great job about Inmoov configuration control center.
It is an optional external thing that will not replace swing or angular dedicated Inmoov interface, if one day someone want to work on it.
Because what we make about a homemade protocol was fun but not strong enough, we need to find a good protocol to communicate between mrl and the server.
We talked about existing mqtt service. I think it is a good option.
I don’t know mrl web services, maybe there is some interesting things too !
Let’s talk about guys

sebastien86, 2017-05-21:
Thank you anthony :-)
Just to help all of you understand what we try to do, you will find in the attach pdf a view of the architecture and 2 mock up screens of the Web interface that could help you to configure motors and parameters of the services you want to use on your inMoov.
All comments, ideas are very welcome.
[python-gui-big-pictures-v1.pdf](https://github.com/MyRobotLab/inmoov/files/1016748/python-gui-big-pictures-v1.pdf)

supertick, 2017-05-21:
This is **Very** Important !
I like your diagram and will review it in detail.  
I have been working for a while on refactoring MRL API - http://myrobotlab.org/content/myrobotlab-api 
The server has always worked with this API, but I have not written a websocket client until recently.
It has worked quite well, but I needed to add some work.  I will have more documentation & code soon.

I'm creating a WebGuiClient (websocket) - and this will refine the API so that other languages and clients can easily implement their own services and borg with remote mrl(s)

There are many schemes and protocols and serializations - each has pros and cons.
Ajax, REST HTTP(S), MQTT(S), TCP, UDP, WebSockets, Json serialization, binary serialization, etc. yatta yatta...

I think the work you have done is **great**, and look forward to happy Borg'ing !

It's all connected !

supertick, 2017-05-21:
I'm curious why the InMoov website should be a dependency ?
Wouldn't it be better to have anyone's InMoov be able to provide an interface for its own configuration ?
This would just involve merging in your current python & html (REST or Websockets) into the build.
What do you think ?

sebastien86, 2017-05-23:
Hi Grog,  
I’m a very end user of MRL and very impress by all the technology you integrate in MRL. 
It’s a great tools with a bunch of possibilities that’s a chance for Robotic  projects because there is very few other open source alternatives.
May be I’m wrong but what I understand is that we can access to the WebServices from the “Home Network” or locally on the computer running MRL. But how to reach the Web Service outside local Network without reconfiguring the NAT rules of the firewall/internet Box ?
The easy way we found to do this is to offer an additional secure Web page to the InMoov users already registered on the InMoov Site, and keep a secure backup of their configuration. Their InMoov account ID is the link with their inMoov Robot. In Case of full scratch of the InMoov computer, just need to reinstall MRL, set the ID and it will recover the config (nice if you don’t have a backup)
Another pro is that if there is a change or new parameter, it will be available for all and automatically updated, with no risk to scratch the existing config files.
This architecture allow also an easy access to basic remote control  functions like sending gestures or type a message/sound to be speak by the robot, just need a tablet and an internet access.
If we do this locally we will lose that kind of end user friendly service.
Tell me if i'm wrong ?

moz4r, 2017-05-23:
Backup config is a great idea.
A "shared gestures" functionality will be great ! Example if I want to send a website community friend "custom happy birthday gesture". 
Or maybe some cool stuffs like avatar video conferences. Endless. You will have some things to do seb :) thank for your contributions, help is needed! 
I agree Grog, we need a swing ( or/and webgui ) thing to configure Inmoov locally.
For people who don’t want edit files / scripting or using this webserver. 
I can work on it , if no one do it, and if I have some time. The first thing maybe is trying to load some config files Inside Inmoov service based on @kwatters work and add some cool buttons

supertick, 2017-05-23:
First I want to say what great work both of you have done. @sebastien86 Your configuration pages (and the whole InMoov site) - looks amazing.  And @moz4r your code-Fu is most impressive, I like the way you jump between Python and now Java - this is the way to be.  To understand it beyond the language its written in.

MRL's network design has always been peer-to-peer.  Where a "peer" in the public ip range would allow spanning of any NAT.  I'm certainly not adverse to the idea of having a separate service on a website which users can benefit from, but there is something very satisfying about running "self contained" software.  Where it can provide configuration locally as well as from the InMoov site.

MRL already has a modular architecture for handling services.  It has 2 UI frameworks, so why not utilize them, and bundle this configuration as another service .. InMoovConfig perhaps?  Potentially, this would allow the peer to peer network to exist with a "peer" mrl instance running on the InMoov site - and distributing config over NATs as was your original intention.

I just want to bring this up as "seeds of thought", I certainly don't want to discourage any of your plans or work.  But at some point in order to distribute this config, you'll need communicate to mrl instances to reload config, or update, or save files, or make backups, and to do so effectively you'll need to use the mrl api.  It is my intention to make this API as easy, and transparent as possible.

Right now I'm working on refactoring some of the API and creating a WebSocketClient.  The WebGui currently uses bi-directional websockets for communication to & from the Angular UI.  A MRL WebSocketClient could provide another asynchronous gateway of messages spanning  NATs.  

These are just some ideas, but I welcome and encourage any ideas from you two very skilled roboteurs ;)

kwatters, 2017-05-24:
I admit my French is limited, but It would be fantastic if we could set up a time to meeting and talk about this integration.  perhaps on a google hangout or something?

I added some very simple methods to the InMoov service, partly inspired by what you've been working on @sbastien86 and @moz4r ..  

I know what I did doesn't line up exactly with what you guys have been doing, but I think they are working towards the same goal.  The rest api's in MRL expose all methods on all services that run in MRL, so, everything is possible.

as for reaching mrl outside of a NAT based network/environment, you are right, we need a centralized relay to so something like this.  If you can proxy requests to your remote MRL instance, then you can control it.   The question is ...

How, as a community, shall we proxy requests to an InMoov from the web to reach our robots remotely even though they are behind a firewall in our own private home? ....

So, let's think differently about it..  maybe it's more like, how do we register our InMoovs with a site that lets us log in remotely and control them.  the InMoovs will call home... just like E.T. ... and we can talk to the robots when they call home..

Ok..  I'm rambling... but I often do that...  I think it'd be great have a google hangout session or something like that to talk about the design and integration...  

Maybe people would even like to listen / join in!

supertick, 2017-05-24:
[image]

This is how distributed MRL looks.  Websocket MRL Messaging API (the same protocol which supports the 'regular' webgui is used to attach (N) number of MRL instances. "All" of these instances can potentially be reached from the public ip if security allows.  Potentially it would also allow messaging from InMoov #1 to InMoov #2 (spanning NATs)

supertick, 2017-08-27:
Keeping ticket - we will get there, but removing from Manticore release

moz4r, 2018-03-06:
Just reread this, maybe we went too far in deep space ? 
based on this maybe we need to think again of something fun and simple to plug on.
For now inmoov swing need some attention.