MyRobotLab/InMoov #79: Some gestures need to deactivate enableAutoDisable

hairygael, 2017-05-20:
For exemple, Da Vinci, I have now made it that it does automatic 
enableAutoDisable=0
The problem is that nothing tells to go back into 
enableAutoDisable=1

moz4r, 2017-05-20:
We need to think a solution , when robot need to be energized or not after an action, this is a very good mind-breaker !

forward, a moveTo blocking action will help, this is an other story. we have puzzle pieces with velocity work.

At this time, a timer will help. exemple by a new command at beginning of a gesture.

i01.forceElectrize(30) >> never enable "autodisable" the servo since 30 seconds. After 30 seconds : 

set autoenable to True again ( if user had set it to true before )

The thing is , if there no new commands after 30s, servo will not de-energized. Not a big issue. the pir or a vocal command will shutdown all to avoid fire emergency

what do you think

hairygael, 2017-05-20:
I'm about to dismantle InMoov, tomorrow early I'm leaving to Bucarest for two conferences.
I found a strange behavior regarding fullspeed and fullSpeed. Hopefully that is fixed.
I tried to set this, but it's very odd. The gesture is being done twice. I don't see what is the issue.
def daVinci():
  i01.disableRobotCanMoveHeadRandom(30)
  i01.rightArm.enableAutoDisable(0)
  i01.leftArm.enableAutoDisable(0)
  i01.setHandSpeed("left", 1.0, 1.0, 1.0, 1.0, 1.0, 0.65)
  i01.setHandSpeed("right", 1.0, 1.0, 1.0, 1.0, 1.0, 0.65)
  i01.setArmSpeed("left", 0.80, 0.80, 0.80, 0.80)
  i01.setArmSpeed("right", 0.80, 0.80, 0.80, 0.80)
  i01.setHeadSpeed(0.75, 0.75)
  i01.moveHead(80,90)
  i01.moveArm("left",0,118,29,74)
  i01.moveArm("right",0,118,29,74)
  i01.moveHand("left",50,40,30,20,10,47)
  i01.moveHand("right",50,40,30,20,10,137)
  sleep(8)
  rest()
  sleep(2)
  i01.rightArm.enableAutoDisable(1)
  i01.leftArm.enableAutoDisable(1)
  i01.rightArm.disable()
  i01.leftArm.disable()

moz4r, 2017-05-25:
there was a conflict with some minimal commands and chatbot , not anymore.
You can now use a new command to electrize servo little more than usual : leftArm.omoplate.forceElectrize(30) > force electrize for 30 seconds

hairygael, 2017-05-27:
Thanks Anthony for the answer.
I will check this out when I return from Romania.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-27:
romania language pack ! find a romanian translator :)
pleasant stay, I never going here, maybe one day

hairygael, 2017-05-28:
Hello Anthony,
Thanks!
I tried setting InMoov for some demonstration to Romanian public using
latest version of MRL. even though we had a great internet connection,
WebkitRecognition gives too much trouble to process, and adds too much
delay for to get responses. The users get confused because the robot
doesn't answer, so they speak over what they already said making
WebKitRecognition stuck to find a correct meaning for a lengthy sentence.

This is actually really an issue which we do not have when using Sphinx.
The response of the robot is rather immediate.
It is the reason I keep using 1.0.107 when doing public demonstrations....

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-05-28:
you can try the fresh new experimental option ( not fully tested )

[image]

if you are happy it can be default option

hairygael, 2017-05-28:
Thanks Anthony!
That sounds like a great experimental option!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2017-06-15 by moz4r