MyRobotLab/InMoov #80: OpenWeather conflict

hairygael, 2017-05-30:
Hello weather man!
English aiml set used to work very nice for weather.
We could ask any city weather in celsius or in farenheit and it would give various answers related with the actual weather of the desired location. It was handy for demonstrations when I was in other countries..
Now with English version, it just says some of the <li> options without giving the weather.
:(

moz4r, 2017-05-30:
mmm conflic with sraix, you have 2 options
- delete openweathermap english chatbot entries related, or rename it
- using openweathermap mrl service by setting the api key >service_H_OpenWeatherMap.config
You can get a free key here : https://home.openweathermap.org/api_keys

[image]

hairygael, 2017-05-30:
Okay, I just tried to remove weathermap in _inmoovChatbot and the conflict is gone.
I will rename the weather map srai to something different to keep the option.

hairygael, 2017-05-30:
Modified the word "weather" to "forecast". Conflict is gone and both can be used.

closed 2017-05-30 by hairygael