MyRobotLab/InMoov #81: jaw not moving in Russian

hairygael, 2017-05-31:
Certainly an issue related to the alphabet.

Calamity01, 2017-05-31:
yes certainly... mouthcontrol is looking for voyels (a,e,i...) to form some
kind of even to open/close the mouth. If those voyels are not present, it
don't know what to do and don't open the mouth

moz4r, 2017-05-31:
ohh I was thinking I had hardware problem ! pushed some russian vowels and french accent too

hairygael, 2017-05-31:
Thanks Calamity for the info!
@Anthony is it working now with what you pushed?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-06-02:
yes it is worky now

hairygael, 2017-06-02:
Thanks for the response.
I'm getting InMoov ready for Madrid, Russian is not what I need now, ehehe!
I will check if it works in Russian anyway and let you know.

moz4r, 2017-06-02:
The day you go inside international space station, please let me know !

hairygael, 2017-06-02:
Tested with version 1.0.2226 and latest inmoov-develop. Mouth is moving once all launching is done. That's very good.
Gee, VoiceRSS is very slow to fetch the mp3. But once cached it's okay.

closed 2017-06-06 by moz4r