MyRobotLab/InMoov #82: jaw and eyes do not autodisable

hairygael, 2017-05-31:
Version 1.0.2205, jaw and eyes do not autodisable even when set to "True" in the head.config.

moz4r, 2017-05-31:
By default if velocity = -1 ( max velocity ) detach action occur after 10 second .
No detach after 10 seconds ?
jaw and eyes by default use max velocity

hairygael, 2017-05-31:
Hello Anthony,
I have in my head.config:
[MAX_VELOCITY]
neck=-1
rothead=-1
jaw=-1
eyeX=-1
eyeY=-1
rollneck=-1

[SERVO_AUTO_DISABLE]
jaw=True
eyeX=True
eyeY=True
neck=True
rothead=True
rollneck=True

When I launch mrl after the init is complete, the jaw disable it self after
about 10 seconds, same for the eyes. But once he has spoken and moved, it
keeps buzzing, and the eyes and jaw are constantly powered.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2017-06-16:
Hello Anthony,
I downloaded the latest MRL version 2247 and latest inmoov-develop, the problem remains.

moz4r, 2017-07-15:
can you send me your inmoov folder ? or better whole mrl folder. because I didn't isolated the problem

hairygael, 2017-07-19:
Hello Anthony,
Testing with version 1.0.2340 with latest inmoov-devellop on InMoov03 and
getting the same result. (Yes Iam printing a third InMoov)
I had not changed anything in the directory inmoov-develop besides:

   - inmoov.config

           -[MAIN]
>             ScriptType=LeftSide
>

   - head.config

           -[MAIN]
>             isHeadActivated=True
>
>             [ROLLNECKSERVO]
>             isrollneckactivated = True
>             RollNeckArduino=right
>             ;RollNeckArduino=left or right
>
>             [SERVO_MINIMUM_MAP_OUTPUT]
>             ;your servo minimal limits
>             jaw=74
>             eyeX=82
>             eyeY=67
>             neck=20
>             rothead=30
>             rollneck=20
>
>             [SERVO_MAXIMUM_MAP_OUTPUT]
>             ;your servo maximal limits
>             jaw=110
>             eyeX=127
>             eyeY=108
>             neck=160
>             rothead=150
>
               rollneck=170

   - arduino.config

           -;my leftport if used
>              MyLeftPort=COM41
>

Trying all kinds of solution,
I just tried and have set to False:

;ramdom move the head
> [MOVEHEADRANDOM]
> RobotCanMoveHeadWhileSpeaking=False
>

And now the jaw is disabling correctly while the eyes are still moving
while speaking and then the eyes are disabling after moving.
Maybe this will give you a hint to find the problem.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-07-20:
Hi dude thank for debug ! It show me the path. I tell you soon results.
Wahooo welcome to I03 !! Big family, you will get a lot of children benefits :)
Look forward to see what upgrade you put Inside I03

moz4r, 2017-07-20:
First step, added visual control Inside servo gui

[image]

moz4r, 2017-07-20:
fixed, pushed

hairygael, 2017-07-24:
Great!
I haven't tested yet, and will take at least 2 weeks before I can have access to my InMoov.
But when ever I can I will let you know how this behaves.

closed 2017-09-08 by moz4r