MyRobotLab/InMoov #84: Head says no while speaking

hairygael, 2017-06-01:
For some odd reason the head moves in a NO movement when he is speaking. And sometimes in a YES movement while speaking. It seems to be set as random.
I think to have a random head movement when he speaks it's nice if desired in the gesture, but this not fitting the correct purpose.
:)
NO head movement when he says "yes sure", isn't proper.

hairygael, 2017-06-01:
Ok I see that you modified Yes and No and I realize that you added if "yes" in Text==Yes()
That explain some of this behavior.
I added yesterday a category with random answers:
<category>
/      <pattern>NICE</pattern>
/      <template>
/        <random>
/          <li>I know </li>
/          <li>yes, indeed </li>
/          <li>you are damn right </li>
/        </random>
/       </template>
/    </category>
If he answers "yes, indeed" now I understand why he moves the head using Yes().
I'm wondering which gesture is triggered when he answers " I know". He is moving his head sideway ,tilting which is okay, but why is he doing that?

moz4r, 2017-06-01:
there was a mistake , about yes and no .
Before, a simple "no" Inside a word was parsed.
Now, we detect only "no " ( a space ) or single word "no" . to be sure it is a real "no"

closed 2017-06-06 by moz4r