MyRobotLab/InMoov #86: Error related i01.RobotIsStarted

hairygael, 2017-06-01:
-Traceback (most recent call last):  File string, line 34, in module  File InMoov/system/InitCheckup.py, line 126, in module    sleepModeWakeUp()  File C:/Myrobotlab/myrobotlab.1.0.2205/InMoov/life\sleepMode.py, line 17, in sleepModeWakeUp    if i01.RobotIsStarted:AttributeError: org.myrobotlab.service.InMoov object has no attribute

moz4r, 2017-06-01:
If I remember i01.RobotIsStarted added Inside java land after 2205 
worky with last build

hairygael, 2017-06-01:
aaah that must it, I'm running 2205.

closed 2017-06-02 by moz4r

vincent, 2017-08-07:
Hi, sorry to revive this thread.

I have the same kind of errors I think, with both latest InMoov code and MRL jar (dev branchs)
I'm not sure I understand "added Inside java land"

Can you point me which versions should I start with, please ?

```
AttributeError: 'org.myrobotlab.service.InMoov' object has no attribute 'RobotIsStarted'
AttributeError: 'org.myrobotlab.service.Arduino' object has no attribute 'usedByInmoov'
```

moz4r, 2017-08-07:
Hello Vincent, thank for your bug report. Inmoov script is now merged Inside MRL . 
Your error say you are using an old MRL version. Here is the solution, the last mrl version : http://34.201.4.170/deploy/develop/myrobotlab.jar

moz4r, 2017-08-07:
( this is last beta version )