MyRobotLab/InMoov #88: predicates AIML not created with the name

hairygael, 2017-06-02:
Hello Anthony,
It seems that default.predicates stays even after giving my name.
Normally a gael.predicates should be created but for some reason it doesn't do it anymore.
Do you have the same behavior?

moz4r, 2017-06-02:
hi gael, I love this place, simple question for head breaker solution ! 
At this this time we can change default username by name gived, but we can look further.
Multiple chatbot session based on facerecognition detection, by exemple. 
I know Kevin has spoken aout similar thing one day, if he read this maybe he can speak up .

hairygael, 2017-06-03:
That would be great of course!
Some one had done something that would do that here:
https://github.com/MyRobotLab/pyrobotlab/blob/master/home/steveh110/opencv.py

kwatters, 2017-06-03:
cool script that steveh110 did!  I think he has it just right.
The important line is:

chatBot.startSession("ProgramAB",nom,"steve")

https://github.com/MyRobotLab/pyrobotlab/blob/master/home/steveh110/opencv.py#L53

The args for that method are 
1. the path for the bots aiml  (normally ProgramAB, this refers to mrl/ProgramAB directory.
2. the name of the person that is chatting with the bot.
3. the name of the bot  so it knows where to load the aiml   ProgramAB/bots/steve in this example.  
 
The important part is that when the face recognizer publishes the face that was detected, you want to start a session with the chatbot with that persons name.

The "startSession" call on ProgramAB should be used to switch the current session for the chatbot.   Which predicates file is being used.  The predicates file should be specific per session. 

When programab starts chatting with a new person  "nom"  it will create a new predicates file for that person by making a copy of the default.properties file.  The idea is, that the GET/SET tags in AIML update the values that are stored in the predicates file on a per user basis.  This way, programab can learn to chat with many people at the same time.

It's been a while since I have tested this stuff, but there are decent tests for ProgramAB.. so hopefully it's still worky.  

So, we want to make this behavior easier...  the reality is the FaceeRecognizer opencv filter will publish the name of the recognized face  (when it detects a new person.)  

So, we could make this easier if we had a chatbot.attach(eyes)   :)    ...  I think this is a really good idea..  I'll see if I can get that added to MRL.. so this will just be easier & automatic..

On a much larger topic,  I'm working on integrating new ways to do face recognition (and general object recognition) using deeplearning4j..  so going forward, I think we're going to want to let the chatbot (programab) know not only about the faces that are recognized, but also other objects that are recognized in a scene...   

fun stuff!  I'm really happy to see all of this progress of pulling everything together like this!

hairygael, 2017-06-03:
Woouii !! Since I have read that you started to implement deeplearning4 into mrl, I can't wait to see it work!
I guess before all, OpenCv should be fixed. Filters keep disappearing after launching face detect or track.
Face recognition is working with version 1.0.2226

kwatters, 2017-06-03:
so.. I made a small change to face recognition this morning...  I should write a blog post and let people know about it.  Now... in the "training" directory, there is a subdirectory for each person.  This is hopefully a pretty minor change, but it does mean that now the image files should be in subdirectories...
So you'd have the top level directory for "training" like it was before.. then you'd have a sub directory for "gael" and your face images would be in that directory..  a different subdirectory for "kevin" and my face images would be in that directory.  

I did this because this is in line with what DL4J is expecting and I'd like to start by using DL4J as an alternative classifier for the face recognition filter.    I think this will also help keep it a bit more organized...

hairygael, 2017-06-03:
Very cool!
That will be a very important step when the robot will be able to get into
personalised sessions.
Subdirectory will be named "gael" and "kevin"?
Do you think there will be a way to start training without manually
clicking the "train" button?
Should the train images be stored on location or should we set a cloud?
That would be amazing to have an inmoov recognizing you anywhere in the
world.
Freaky too!!

Just to say, 1.0.2226, everything is worky currently with InMoov except
face tracking. Do you think we could focus on OpenCV to get a MRL version
released and stable before we modify fun stuff?
:)
We are so close that I freak to the idea that stuff might be broken again.

So many members are turning to EZ robot, it's too bad.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2017-07-10:
Woouiii, hello guys!
Since a few versions I cannot get inmoov to remember my age. And if I ask his age, I get a NULL ERROR RESPONSE from programAB or even it freeze sometimes MRL.
The bug is a tricky one because when using InMoov in public, one of the first people's question is : How old are you? 
I have been checking the AIML to see if something  has been changed, but can't find anything recent. I suspect this could be related to saving the predicates.
Any idea?

hairygael, 2017-07-10:
Okay I tried with modifying utilities.aiml  to TRUE and it works. I wonder what else has been changed that made this not working.
```
<category>
    <pattern>ISANUMBER *</pattern>
    <template>true</template>
  </category>
```

moz4r, 2017-07-10:
Hi Gael ! 
I saw this file it seem there is important informations missing. Some <set></set> tags are gone. I will repair.

moz4r, 2017-07-10:
fixed, soon no age problem anymore, singularity is coming :)

hairygael, 2017-07-10:
Nice I will test it.
so I can set back to false?
```
<category>
    <pattern>ISANUMBER *</pattern>
    <template>false</template>
  </category>
```
When asking to the robot "how old are you" is also fixed?

hairygael, 2017-07-10:
Just tested, it's worky great!
I need to keep the "true" set up

moz4r, 2017-07-10:
great ! ISANUMBER can be used elsewhere . the correct syntaxe is : 

`
<category>
--
  | <pattern>ISANUMBER <set>number</set></pattern>
  | <template>true</template>
  | </category>

  | <category>
  | <pattern>ISANUMBER *</pattern>
  | <template>false</template>
  | </category>

`

moz4r, 2017-10-24:
To cook : 
- Tracking + Facerecognizer ( if worky )
- Unknown face : store to temp / check temp > ask the name > learn things
- Know face : say hello / start personal session

time to cook

moz4r, 2017-11-05:
Pushed , worky ( for me, for you too ? )

https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/services/D_OpenCv.py#L118

hairygael, 2017-11-06:
> Pushed , worky ( for me, for you too ? )
>
> https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/
> services/D_OpenCv.py#L118

I will test it tomorrow and let you know!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-11-16:
we can close it ?

hairygael, 2017-11-16:
Yes it's fixed

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2017-11-16 by moz4r