MyRobotLab/InMoov #90: русский язык

moz4r, 2017-06-12:
Так как нам не удалось поработать вместе вол лог файл: https://yadi.sk/d/z09zVxzJ3JkPgQ
По моему там ошибки с распознаванием кодировок для русского языка. В googlehrom при голосовых командах ”камень ножницы бумага” есть какая то реакция на все остальные команды в ответ перечисление цифр и символов на русском языке но нет никакой логики. Если нужно я готов предоставить свой WINDOWS 7 для доступа только назовите число и время.Спасибо.

moz4r, 2017-06-14:
fixed
https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/chatbot/bots/ru/aiml/_inmoovApps.aiml#L60

AlexInMoov, 2017-08-14:
Приветствую Вас. Вот файлы "aiml" исправленные для русского языка: https://yadi.sk/d/0s7GD9gN3LxKgg
вот файлы "Gestures" для поддержки разных языков:  https://yadi.sk/d/FhFskLnW3LxSbd
но в них нужно исправить файлы:  https://yadi.sk/d/1a3a5Prw3LxQAT
вот файлы "InMoov\system\languagePack\ru" исправленные для русского языка: https://yadi.sk/d/UpRB8kwG3LxGFP
спасибо за сотрудничество.

moz4r, 2017-08-17:
Hi Alex ! Thank you for translation. we will push it. 
I need some time to globalize gestures files. I tell you soon

moz4r, 2017-08-17:
Hi @hairygael what do you think about this gesture model to implement Alex traduction work ? https://github.com/MyRobotLab/inmoov/blob/7917c08815924493a8d54fcc02a77f9b0d265f6f/InMoov/gestures/about.py

hairygael, 2017-08-17:
> 9b0d265f6f/InMoov/gestures/about.py

Hello Anthony,
Yes that is nice, this way each gesture could contain the translated
version.
Good progress!!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-08-21:
while I convert things, I was thinking of something more sexy :

i01.speakBlocking("FR","bonjour mon nom est Inmoov")
i01.speakBlocking("EN","hello my name is Inmoov")

hairygael, 2017-08-21:
Great that's even cleaner!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

AlexInMoov, 2017-08-23:
Приветствую Вас moz4r и hairygael. Вот файлы для русского языка исправленные для правильного произношения" learnf" -   https://yadi.sk/d/-tfDbCbn3MGqhk
"_inmoovApps.aiml" -   https://yadi.sk/d/NdgYQraA3MGqjW
 " InMoovApps  " -     https://yadi.sk/d/A6AvNHks3MGqqi

kwatters, 2017-08-23:
I think the proper thing would be to not have any of the strings in the gesture python scripts at all.  In reality, there should be a file that defines all of the unique strings that you can say.. and those should be referenced from the gestures.

This is how localization usually happens in applications.  
So, you'd have a python file for each language you want to support

Something like  a file called "utterances_en.py"   this file would be sourced prior to loading the gestures.

text_data[1] = "I am the first life size humanoid robot you can 3D print and animate"
text_data[2] = "blah blah blah" 

Then in the gestures file like  about.py you should have something that references it like

i01.speakBlocking( text_data[1] )  

to change the language you are working with , at that point would just be to change which "utterance" file you load.

I don't like the idea of having speakBlocking("EN", "some stuff" )  It's confusing....and generally not the right way to do it.

moz4r, 2017-08-23:
Hi Kevin ! We have set all the system on this model, one file to store all languages utterances.

https://github.com/MyRobotLab/inmoov/tree/develop/InMoov/system/languagePack/en
https://github.com/MyRobotLab/inmoov/tree/develop/InMoov/system/languagePack/fr
...

But I'm afraid to do the same thing Inside gestures files. This can be strange for users who want to script.
Inside one uniq file, it is easy to script language+moveTo.
i01.speakBlocking( text_data[X] ) maybe hard to debug

why not...
lets talk about, to find the best way for all

waiting to push last mods

kwatters, 2017-08-23:
it doesn't need to be text_data[x]   .. it can be something more descriptive..   the approach of having general.lang is a good one.  It should be a python compliant syntax for that file and it should be used consistently.  

I don't want to see the speech interface now have to support something like  speakBlocking(language, text)  

The speech synthesis interface already defines the setLanguage() method

https://github.com/MyRobotLab/myrobotlab/blob/develop/src/org/myrobotlab/service/interfaces/SpeechSynthesis.java#L24

I see that you've also added it to the main inmoov service.  I don't think it should be in both places..  Having 2 places to configure the same thing is a good recipe for bugs and confused users!

So,  I'm worried about some of these changes because it's adding new code for stuff that already exists.

Another option is a mapping file that maps from the original english text to the translated text.. again this should be stored as a separate file and not in the python code.

So..  you could have a mapping file that has a line like 
utternace_fr.py

translationMap = {}
translationMap['hello']= u'bonjour'

And then if the speech service has a handle to this map, and it knows it needs to speak in french  (because you called  setLangauge on the mouth)  it would look at what it's being asked to say, look it up in the translation map .. and use that value instead..  

I'd like to hear from more MRL folks on this one..

moz4r, 2017-08-26:
So lets refactor gestures !
But this will be done after manticore, because some work is needed and other discussions. 
At this time I will push translation as comments. waiting the day of gestures refactor.
As bonus, some kind of basic GUI for gesture edition will help a lot.
Global inmoov GUI is an other subject, this can be a swingGui tab, friend/mod of gesture creation tool.
So translationMap or text_data[x] will be more fun.

moz4r, 2017-08-27:
@AlexInMoov files are pushed inside last testing version

closed 2017-09-08 by moz4r

AlexInMoov, 2018-02-13:
Доброго времени суток! moz4r как на счет Gestures с поддержкой русского языка. Требуется отредактировать код чтоб с каждым выпуском новой версии на редактировать Gestures. Занимает много времени. Необходимо определиться с алгоритмом и мы готовы помочь с редактированием. Дайте шаблон и мы поможем отредактировать все файлы.