MyRobotLab/InMoov #92: ProgramAB fixes

moz4r, 2017-07-03:
Todo

Fix aiml/csv synchronisation and remove "delete aimlif trick"
Push last programAB repo about Kevin fixes ( utf8 )
Integrate learn tag Inside french chatbot

moz4r, 2017-07-06:
some work to do inside french chatbot but core seem WORKY

moz4r, 2017-07-07:
todo : UTF8 > writeCertainIFCaegories

kwatters, 2017-07-07:
is the writeCertainIFCategories method not utf-8 safe?    I made some changes to programab to support utf-8 in some cases, but I haven't merged this updated version to MRL yet.   

https://github.com/kwatters/program-ab/commit/932d1fa92bef8379aa16def365482e26b60d13e8

I think all of these utf-8 fixes need to be done in program ab, not MRL.

moz4r, 2017-07-07:
Hi kevin,  I have merged you previous work as  0.0.2-kw inside programAb service. worky great !
I suspect writeCertainIFCaegories not utf8 friendly, I can try to fix

kwatters, 2017-07-07:
Awesome!  I didn't realize you merged it in, I never had a chance to test it.  I'm glad to hear it fixed the issue.  I'll update the writeCertainIFCategories ..  (it's actually the underlying method writeIFCategories that isn't utf-8 safe.  )   I'll check it in as 0.0.3-kw and add to the mrl repo.

kwatters, 2017-07-07:
I've updated that message and bumped the version to 0.0.3. I have not updated the ProgramAB service in MRL., but I did add it to the repo.

https://github.com/kwatters/program-ab/commit/224c122e4e9febfde56a4aa2d2d560e217e80962#diff-1baeabaa922cfa70821b2ccb153942dbR382

moz4r, 2017-07-10:
updated, we see the light, it is almost perfect! There is still something that corrupt all the set by converting utf8 to bad symbols. I'm trying to find it, for kill it. not found at this time

moz4r, 2017-07-10:
I think it killed yehhh until next one ;) can you merge ? I cant

kwatters, 2017-07-10:
Merged!

moz4r, 2017-07-15:
seem worky great !

closed 2017-07-15 by moz4r