MyRobotLab/InMoov #95: Loop in START_INMOOV.bat / InMoov.bat

MatsMRL, 2017-07-25:
This script: https://raw.githubusercontent.com/MyRobotLab/inmoov/develop/START_INMOOV.bat
seems to create a loop. It launches the install over and over again. 
I tested with the a the 2359 build.

moz4r, 2017-07-26:
Hi mats thank for info. Are you using public jar or eclipse build ?

MatsMRL, 2017-07-26:
The public jar.

I tried to follow the instructions from here:
https://github.com/MyRobotLab/inmoov/wiki/HOWTO---SETUP-&-PREREQUISITES
Step by step as a beginner would do.
So I think the problem only occurs when downloading a fresh myrobotlab.jar
and then executing this .bat file
https://raw.githubusercontent.com/MyRobotLab/inmoov/develop/START_INMOOV.bat

moz4r, 2017-07-27:
I could not reproduce this issue. strange. I will try again

MatsMRL, 2017-07-28:
OK.

It works fine when using
java -jar myrobotlab.jar -install
and then using the START_INMOOV.bat

Seems like Luis has made some progress:
https://groups.google.com/forum/#!topic/inmoov/zs9ReomPUug

/Mats

moz4r, 2017-07-28:
Found 2 problems this afternoon.
- No more cofee
- If there is "spaces" or complex folder like c:\users\spa cé\Documents\mrl, system get crazy.

Fixed now ( a red warning will occure ). 
If you have some time can you make a new test about loop ?
Great about Luis, after stable mrl release maybe we can package mrl+launcher+mrlcomm for beginners ?

closed 2017-08-28 by moz4r