MyRobotLab/InMoov #97: Inmoov shutdown broken

moz4r, 2017-08-01:
todo : use stopService() from java land

supertick, 2017-08-02:
Runtime.shutdown(sec) not working ?

moz4r, 2017-08-02:
Strange issues from python land. 
The solution to use service related stopService() is cleaner I think

supertick, 2017-08-18:
Just to be clear ..
stopService stops the threads ...
releaseService calls stopService AND purges all scheduled tasks AND removes the service from the registry and publishes events saying the service has been released ..
So if you want to completely remove a service use releaseService

moz4r, 2017-08-19:
Thank you for info ! I will use release when I want to shutdown . At this time I just put some things inside services shutdown routines related. Like writeAndQuit inside programab. Or rest()+disable inside Inmoov. Goal is to run some tasks when user use the swingui cross.

hairygael, 2017-08-19:
Hello guys,
This is more a message for Anthony.
When InMoov is shutting down (inmoov-develop), it gets back to rest()
position, that is good and safe.
But after that, it doesn't disable() the servos before shutting down MRL
completely. (version MRL 1.0.2340)
So, we currently need to manually turn off the power to make sure the
servos stop buzzing.

Is it something fixed with more recent versions or does it still remain?

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-08-27:
just tested the last build on whole robot, seem ok. for you too ?

hairygael, 2017-08-28:
Hello Anthony,
I will not be able to test for a while. My third InMoov is now set on the
exhibition. My second InMoov is dismantled in suitcases ready to go to the
Makerfaire Eindhoven, and my first InMoov is out of the workshop for a show.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-08-28:
aaarg master without InMoov :( 
You need a fourth inmoov !!!
Have fun for Makerfaire

hairygael, 2017-08-28:
Thanks Anthony!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-09-14:
closing - Gael can re-open if noWorky for him ..
Thanks..

closed 2017-09-14 by supertick

hairygael, 2017-09-16:
Shutting down is fixed, very good!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>