MyRobotLab/InMoov #99: InvocationTargetException in MLR 2405

airhic, 2017-08-17:
Hi Dev Team,

Since I use MRL 2405, I receive a huge number of errors like this when chatbot (chrome browser) waits for words:

15:27:54.679 [i01] ERROR c.myrobotlab.framework.Service - InvocationTargetException
java.lang.reflect.InvocationTargetException: null
	at sun.reflect.GeneratedMethodAccessor96.invoke(Unknown Source) ~[na:na]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[na:1.8.0_141]
	at java.lang.reflect.Method.invoke(Method.java:498) ~[na:1.8.0_141]
	at org.myrobotlab.framework.Service.invokeOn(Service.java:1351) [myrobotlab.jar:1.0.2405]
	at org.myrobotlab.framework.Service.invoke(Service.java:1287) [myrobotlab.jar:1.0.2405]
	at org.myrobotlab.framework.Service.run(Service.java:1668) [myrobotlab.jar:1.0.2405]
	at java.lang.Thread.run(Thread.java:748) [na:1.8.0_141]
Caused by: java.lang.NullPointerException: null
	at java.util.concurrent.ConcurrentLinkedQueue.checkNotNull(ConcurrentLinkedQueue.java:920) ~[na:1.8.0_141]
	at java.util.concurrent.ConcurrentLinkedQueue.offer(ConcurrentLinkedQueue.java:327) ~[na:1.8.0_141]
	at java.util.concurrent.ConcurrentLinkedQueue.add(ConcurrentLinkedQueue.java:297) ~[na:1.8.0_141]
	at org.myrobotlab.jme3.InMoov3DApp.setMicro(InMoov3DApp.java:90) ~[myrobotlab.jar:1.0.2405]
	at org.myrobotlab.service.InMoov.onListeningEvent(InMoov.java:2019) ~[myrobotlab.jar:1.0.2405]
	... 7 common frames omitted
...
15:28:02.681 [i01] ERROR c.myrobotlab.framework.Service - InvocationTargetException
java.lang.reflect.InvocationTargetException: null
...
15:28:02.695 [i01] ERROR c.myrobotlab.framework.Service - InvocationTargetException
java.lang.reflect.InvocationTargetException: null
...
15:28:08.524 [i01] ERROR c.myrobotlab.framework.Service - InvocationTargetException
java.lang.reflect.InvocationTargetException: null
...

I never get this in MRL 2386. I use Linux platform.

Thanks in advance for your help

Regards,
Eric

kwatters, 2017-08-17:
You should submit a no worky or attach your myrobotlab.log file to the ticket.  Also please provide some details about what you're actually trying to do.  It sounds like you're starting a chatbot service,  I assume you mean the ProgramAB ..  Based on you saying it's waiting for words, I suspect you're maybe using the WebkitSpeechRecognition service also?

Is this correct?

airhic, 2017-08-17:
Hi Kevin,

I use MRL with InMoov and use the package provided by Gaël/Antony.
https://github.com/MyRobotLab/inmoov/wiki/HOWTO---SETUP-&-PREREQUISITES

I configure _InMoov.config file to use Virtual and french language. In this case, I get the issue and exceptions (same issue with english language). If I close Chrome browser I didn't receive any exception at all. In the chatbot tab, I can continue to chat with InMoov by writting some questions and so on... without any exception.  

I made another test. In _InMoov.config, I've changed ScriptType=Virtual by NoArduino. In this case I didn't receive any exception, chatbot works !

I'm newbie on MRL and InMoov. I suppose that chatbot uses ProgramAB and WebkitSpeechRecognition. I could see that in another config files. 

I'm sorry but version 2386 worked last week but today I get the same behavior ! Don't know why :-(
I add myrobotlab.log file from the working and non working test. I hope this help.
[mrl-log.zip](https://github.com/MyRobotLab/inmoov/files/1232149/mrl-log.zip)

Thanks,
Eric

hairygael, 2017-08-17:
Hello Eric,
Thanks for the bug notification.
I have a set up a ZIP worky version package on InMoov website if you need
it to work.
Download script : myrobotlab.1.0.2340.zip
<http://www.inmoov.fr/wp-content/uploads/mrl/myrobotlab.1.0.2340.zip> and
extract.

Remember that MRL is a work in progress and intermediate versions might
have bugs.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-08-27:
Hi !
can you retry with the last build?

airhic, 2017-08-28:
Hi Antony ! 

This issue is fixed with 2429 build. 
Thanks a lot ! 

Eric

moz4r, 2017-08-28:
Happy to ear that sound ! see you

closed 2017-08-28 by moz4r