MyRobotLab/InMoov #100: Py exception with random life

hairygael, 2017-08-21:
Hello Anthony,
I found a new bug for you, as I'm sure you are lacking of things to do right now.
:)
The bug is, if you do not have all body parts set as True, and call for the command "Random life", you get a Py exception.
Running version MRL 1.0.2340, and set main script as Full instead of Virtual.
But the left hand is not attached so I have set the skeleton left hand config to False.
Running command Random life, gets me a Py exception and random will not work.

moz4r, 2017-08-25:
Hi ! I will check soon

moz4r, 2017-08-28:
soon is today, fixed !

hairygael, 2017-08-28:
Great!
I wish I could test it, but without an InMoov at hand it's difficult.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2017-09-21 by moz4r