MyRobotLab/InMoov #106: wake up

hairygael, 2017-09-06:
Hello testing Version MRL 1.0.2483 with InMoov Master.
When InMoov goes to sleep, the microphone cannot listen. It would be nice if you could vocaly tell him "wake up".
If writting in the chatbot tab: wake up, the robot will wake up but the mic is then still deactivated.
So, the wake up function works properly if the PIR detects movement, and then the mic is correctly working.

Could we fix that easily?

moz4r, 2017-09-06:
easy !

[image]

will look at this . useful too if we don't have pir

moz4r, 2017-09-06:
precision before commit, do you want the microphone always ON while robot sleeping ? ( off course it will awsers only few commands like wake up or shutdown )

moz4r, 2017-09-06:
it can be a personnal config thing..

moz4r, 2017-09-06:
ok pushed inside develop , you can just replace ear.py
robot will respond with microphone or program ab manual entry
will push later this mod inside the jar because quota exceeded :)

hairygael, 2017-10-01:
Tested and worky!! It's really much better!!
Now lets make it still more difficult.
:)
I think as long as we speak to the robot, even if the PIR doesn't detect movement, the robot should not go to sleep, unless we tell him to do so.

moz4r, 2017-10-02:
pushed
robot will no longer be insolent and will not fall asleep when you speak to him.

hairygael, 2017-10-02:
Nice!!
I will test that tomorrow.
:)

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2017-10-05:
This is super!

closed 2017-10-05 by hairygael