MyRobotLab/InMoov #108: InMoov service documentation

moz4r, 2017-09-13:
@hairygael can we start to edit as draft http://myrobotlab.org/service/inmoov
Goal is to refresh links, services names, minimal scripts, full script , todo list, setup ... before the release.

hairygael, 2017-09-13:
Hello Anthony,
Yes sure!
The InMoov service looks a bit outdated by now.
I won't be around for a few days though...

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

supertick, 2017-09-14:
This can be the issue to track.
Closing
[Make pyrobotlab/service/InMoov.py both cool & worky !!!](https://github.com/MyRobotLab/pyrobotlab/issues/84 )
as a duplicate but please be aware - the script is in pyrobotlab repo.

The reference to the **service page** should be through the develop branch like this 
http://myrobotlab.org/service/InMoov?branch=develop
it will be copied into master when we release ...

The service page auto-magically appends the pyrobotlab/service/{serviceName}.py script
https://github.com/MyRobotLab/pyrobotlab/blob/develop/service/InMoov.py

These references will auto-magically get fetched from help, or info, or many other ways.  The script will auto-magically get tested every build, so we will all know quickly if it is borked.  

It's important that this script is
- worky
- easy and clear to configure & tweak
- well commented
- will switch to all virtual hardware when 
> virtual = true
- fun

moz4r, 2017-11-16:
Doc in progress , guys is it the way to go ? http://myrobotlab.org/service/InMoov?branch=develop

moz4r, 2017-11-17:
@hairygael & others maybe we need to remove "ear.addcommand" from full script. It is unused and it's confused. Because chatbot activated by default. 

ear.addcommand are used inside DIY scripts ( also called testScripts / service scripts ) . I think it is enough ?

closed 2017-12-25 by moz4r