MyRobotLab/InMoov #110: possible issue with mrl remoteadapter service

CheekyMonkey, 2017-09-15:
I'm seeing an exception in the mrl log when I attempt to make a remote adapter connection between two Raspberry Pi 3's

The remote adapter connection between the two Pis is failing as a result

[video](https://www.youtube.com/watch?v=EUvBbooTrSI) 

As the mrl GUI freezes after the remote adapter connection failure, I'm not able to send the mrl log through the menu in the mrl GUI so the mrl log from each Raspberry Pi are being shared here for reference

[Raspberry Pi 3 - 1 myrobotlab.log.txt](https://github.com/MyRobotLab/inmoov/files/1306054/Raspberry.Pi.3.-.1.myrobotlab.log.txt)
[Raspberry Pi 3 - 2 myrobotlab.log.txt](https://github.com/MyRobotLab/inmoov/files/1306055/Raspberry.Pi.3.-.2.myrobotlab.log.txt)

supertick, 2017-09-15:
Wow .. this one of the most beautiful bug reports I have ever seen.
Clear, concise, even a Video ! 
Simply awesome.  Thanks !

So a couple of new non-serializable things got put into Runtime & SwingGui - this can make remote adapter connections explode BOOM.

This can be on-going issues as different services grow, so I'm getting the Test service to test this (Test service needs a little refactoring itself).

Fixed.

CheekyMonkey, 2017-09-15:
nice work supertick :)
now confirmed worky :)

[video](https://youtu.be/NPro2f4xZX8)
[image]

Cyber-One, 2017-09-15:
Not so worky once you start to add services.
On remote Raspi,
# Fred01 startup Script

# This script is to  start each of the servo used in the InMoov Robot Fred.
# Fred consists of 3 Raspberry Pi 3.
# This is for the First Raspberry Pi 3 located in Fred's head and has the I2C servo interface card installed.

# There are 4 I2C servo controllers
head = Runtime.createAndStart("head", "Adafruit16CServoDriver")
rightArm = Runtime.createAndStart("rightArm", "Adafruit16CServoDriver")
leftArm = Runtime.createAndStart("leftArm", "Adafruit16CServoDriver")
back = Runtime.createAndStart("back", "Adafruit16CServoDriver")

#  This is all running off a Raspberry Pi 3
raspi = Runtime.createAndStart("raspi","RasPi")

# Now attach the servo drivers to the Raspberry Pi 3.
head.setController("raspi","1","0x40")
rightArm.setController("raspi","1","0x41")
leftArm.setController("raspi","1","0x42")
back.setController("raspi","1","0x43")

# Now lets create the servo functions

# We will Start with the Head
jaw = Runtime.createAndStart("jaw", "Servo")
jaw.attach(head,2)
jaw.setMinMax(0,180)

headX = Runtime.createAndStart("headX", "Servo")
headX.attach(head,3)
headX.setMinMax(0,180)

eyesX = Runtime.createAndStart("eyesX", "Servo")
eyesX.attach(head,0)
eyesX.setMinMax(0,180)

#  Right Arm
rightBicep = Runtime.createAndStart("rightBicep", "Servo")
rightBicep.attach(rightArm,15)
rightBicep.setMinMax(0,180)

rightElbow = Runtime.createAndStart("rightElbow", "Servo")
rightElbow.attach(rightArm,14)
rightElbow.setMinMax(0,180)

rightWrist = Runtime.createAndStart("rightWrist", "Servo")
rightWrist.attach(rightArm,0)
rightWrist.setMinMax(0,180)

rightThumb = Runtime.createAndStart("rightThumb", "Servo")
rightThumb.attach(rightArm,1)
rightThumb.setMinMax(0,180)

rightIndex = Runtime.createAndStart("rightIndex", "Servo")
rightIndex.attach(rightArm,2)
rightIndex.setMinMax(0,180)

rightMajure = Runtime.createAndStart("rightMajure", "Servo")
rightMajure.attach(rightArm,3)
rightMajure.setMinMax(0,180)

rightRing = Runtime.createAndStart("rightRing", "Servo")
rightRing.attach(rightArm,4)
rightRing.setMinMax(0,180)

rightLittle = Runtime.createAndStart("rightLittle", "Servo")
rightLittle.attach(rightArm,5)
rightLittle.setMinMax(0,180)
[... 189 lines cut ...]
09:05:46.280 [RemoteAdapterB:tcp://192.168.1.22:6767] ERROR c.m.n.TcpThread [TcpThread.java:69] shutting down thread
09:05:46.281 [RemoteAdapterB:tcp://192.168.1.22:6767] ERROR c.m.n.TcpThread [TcpThread.java:71] attempting to close streams
09:05:46.283 [RemoteAdapterB] ERROR c.m.f.Service [Service.java:2042] RemoteAdapterB error SocketException - Socket closed java.net.SocketException: Socket closed
	at java.net.SocketOutputStream.socketWrite(Unknown Source)
	at java.net.SocketOutputStream.write(Unknown Source)
	at java.io.ObjectOutputStream$BlockDataOutputStream.drain(Unknown Source)
	at java.io.ObjectOutputStream$BlockDataOutputStream.setBlockDataMode(Unknown Source)
	at java.io.ObjectOutputStream.writeNonProxyDesc(Unknown Source)
	at java.io.ObjectOutputStream.writeClassDesc(Unknown Source)
	at java.io.ObjectOutputStream.writeOrdinaryObject(Unknown Source)
	at java.io.ObjectOutputStream.writeObject0(Unknown Source)
	at java.io.ObjectOutputStream.writeFatalException(Unknown Source)
	at java.io.ObjectOutputStream.writeObject(Unknown Source)
	at org.myrobotlab.net.TcpThread.send(TcpThread.java:308)
	at org.myrobotlab.net.TcpServer.sendTcp(TcpServer.java:156)
	at org.myrobotlab.service.RemoteAdapter.sendRemoteTCP(RemoteAdapter.java:306)
	at org.myrobotlab.service.RemoteAdapter.sendRemote(RemoteAdapter.java:296)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.myrobotlab.framework.Service.invokeOn(Service.java:1351)
	at org.myrobotlab.framework.Service.invoke(Service.java:1287)
	at org.myrobotlab.framework.Service.run(Service.java:1672)
	at java.lang.Thread.run(Unknown Source)

supertick, 2017-09-16:
Thanks Cyber-One ..
I like your startup, in that you fully decomposed all the pieces I assume for the I2C control of the servos.
Bravo !

I'll re-open this ticket and look into it

supertick, 2017-09-16:
Servo was borked with non-transient moveToBlocked lock - fixed in build 2504

closed 2017-09-16 by supertick

CheekyMonkey, 2017-09-16:
Thanks Grog, nice work again :)

Raspberry Pi 3/AdaFruit16C I2C connected servos/Remote Adapter service confirmed worky  with development build 2505

[Video](https://www.youtube.com/watch?v=ZvQ7h4jXdt0&feature=youtu.be)

[image]