MyRobotLab/InMoov #111: i01.disableRobotRandom(30)

hairygael, 2017-09-28:
Hello Anthony,
How can we disable the random while doing a gesture now?
 i01.disableRobotRandom(30) is not attributed anymore, this was added in some gestures which aren't working anymore.
I thought the function startedGesture() and finishedGesture() was going to do that but I guess I misunderstood something.
:)

moz4r, 2017-09-28:
Hi Gael ! i01.startedGesture() will do the job ( + finishedGesture() at the end )

what do i01.startedGesture() : 
- stop servo autodisable
- stop random things
- warn if we launch an other gesture
- we can add other things like stop microphone etc ...

hairygael, 2017-10-01:
Thanks Anthony,
I just tried with version 2552 and it works!
I need to clean up all the gestures that had i01.disableRobotRandom(30)

hairygael, 2017-10-01:
In InMoov/Life/MoveRandomize.py there is still:
i01.disableRobotRandom(20)
Is that okay?

moz4r, 2017-10-02:
bien vu ! 
fixed

hairygael, 2017-10-02:
Great!
Thank you

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2017-10-03 by moz4r