MyRobotLab/InMoov #112: Track voice command is borked

hairygael, 2017-10-01:
Hello, something is not working anymore in java land regarding LKOpticalTrack.
```
def trackPoint():
  i01.headTracking.startLKTracking()
  i01.eyesTracking.startLKTracking()
  i01.setHeadVelocity(80, -1)
  sleep(1)
  fullspeed()
```

In InMoov.java:
 
```
public void trackPoint() {

    if (eyesTracking != null) {
      eyesTracking.startLKTracking();
      eyesTracking.trackPoint(0.5, 0.5);
    }

    if (headTracking != null) {
      headTracking.startLKTracking();
      headTracking.trackPoint(0.5, 0.5);
    }
}
```

moz4r, 2017-10-02:
fixed & pushed

hairygael, 2017-10-02:
Thanks Anthony,
I will test tomorrow.
I see there is a lot more to add now in the script.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

closed 2017-10-08 by moz4r