MyRobotLab/InMoov #113: NewYork gesture solution

hairygael, 2017-10-02:
The NewYork gesture used to be fun for demo.
The way it worked was to launch the song I arranged with a friend of mine with:
i01.mouth.speak("robot1")
[](https://github.com/MyRobotLab/inmoov/blob/develop/InMoov/system/sounds/robot1.mp3)

When the python script would reach: 
i01.mouth.speakBlocking("Start spreading the news")
I had replaced the "Start spreading the news.mp3" created with a 0 byte mp3 in the audiofile directory, therefore the jaw would move but it wouldn't say the words in order to not overlap the real song which already contained the words.

I wonder if we could do something similar again...

moz4r, 2017-10-03:
Pushed something, it is not very clean ... but worky with software mouthcontrol + audiofile + all tts. 

https://www.youtube.com/watch?v=6YZMh44mwrk

3 things : 
- New york gesture : refactored to play the mp3 ( without mouth control ) + pushed this event when we want robot move the jaw ( i01.mouthControl.onStartSpeaking("I want to be a part of it to be a part of it") ). Text is not really important, it's for timing.

- Star wars sound : you can call now whataboutstarwars() without deal with mp3 files. It will worky.

- There is something on the bed table to refactor and standardize realtime mouthcontrol, without any additional hardware. So tomorrow newyork song will be playable with 2 tracks ( 1 for voices + 1 for music ).

moz4r, 2017-10-03:
Just want to say I have this song in the head now ;)

hairygael, 2017-10-03:
Great I need to test it!!
Just to say, I never posted a video on YouTube of my InMoov playing the NewYork song and never allowed medias to film it. I think there might be some right reserved.

moz4r, 2017-10-03:
a video ? where do you see a video :)

hairygael, 2017-10-03:
No video??
Ehehe!
Just tested the Newyork gesture, it feels nice to see it work again!

closed 2017-10-05 by hairygael