MyRobotLab/InMoov #114: Jaw doesn't disable

hairygael, 2017-10-03:
Hello Anthony,
Version 1.0.2563 with latest inmoov-develop.
Jaw doesn't autoDisable during random life and doesn't move while speaking after Random Life.
Also Jaw velocity stays on 25 after stopTracking, so that must be why it doesn't move while speaking.

Relax gesture doesn't autoDisable, but it doesn't seem related to Random life.

moz4r, 2017-10-05:
some little problems but you put a finger on a bug well hidden for a long time. ( last servo position sometime bad ). 
fixed. Is it better ?

hairygael, 2017-10-05:
It seems to be good now for the jaw!

Relax is not okay, removing the i01.startedGesture() solves the autodisable issue but stops the gesture from being complete.

moz4r, 2017-10-05:
I think it is the same issue related to last events not blocked, we need more magic : 

https://github.com/MyRobotLab/inmoov/issues/116

closed 2017-10-08 by moz4r