MyRobotLab/InMoov #115: Questions

hairygael, 2017-10-05:
I have a question (Actually more than one, but lets go step by step)
-When inmoov is on random life, the head doesn't move while speaking, is that possible to change?
-When in Random life could it be possible if the body moves that the head moves as well? It's looking not very human when only the body moves and the head stays still.

moz4r, 2017-10-06:
[image]

Yeh we can enhance things about skeleton random things.
There is a new parameter that activate tracking if pir detect something, do you want we cook it inside random life too ? of course random head is disabled if tracking activated. Even while speaking.

@Calamity01 started some time ago to work on servo acceleration. servo acceleration / deceleration will be great tool for smoother things. Don't know more about it at this time.

will work on more human things

hairygael, 2017-10-11:
Thanks Anthony,
Yes I have seen the tracking if PIR detects something. Nice option!
I currently have turned it off, because the fps is still dropping too low
due to OpenCV 3.0.

Ramping for accelerations and decelerations is something that we talked
about already 5 years ago with Marten and Grog.

If we can set the head to random if body is set on random for now it would
be very nice, but there is no hurry.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

moz4r, 2017-10-21:
Pushed, but I think we need to keep this post opened "ad vitam aeternam" because there is so much to do about "natural" reaction

hairygael, 2017-10-24:
Yes right!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>