MyRobotLab/InMoov #116: Conflict between startedGesture and moveHeadWhileSpeaking

hairygael, 2017-10-05:
You can reproduce this conflict when telling to the robot: look on your left side.
He will not behave always the same.
Sometimes he will turn the head correctly to the left side, but sometimes he will move the head randomly instead of turning the head to the left side.
It must be a lack of delay between the two commands.
Version mrl 1.0.2563, last inmoov-develop

But I also wonder if startedGesture() doesn't have an issue now, why doesn't it autodisable after finishedGesture?

hairygael, 2017-10-05:
Hello Anthony,
i01.moveHeadBlocking(85,160) has no attribute. I just tested.
Unfortunately, I need to dismantle InMoov now, because tomorrow morning, I am taking the plane.
So I won't be able to test for a few days.

Version of MRL 1.0.2251 was working okay though with startedGesture on Relax and all the other gestures. The autodisable was doing okay after finishedGesture.

moz4r, 2017-10-05:
Hi gael , oh I just forgot to say global i01.moveXBlocking is not yet implemented :)
it is just on the todo list ( near the top ) 
Have a nice trip over the clouds !

hairygael, 2017-10-05:
Ah ok understood!
Thanks.

moz4r, 2017-10-17:
Pushed , 

Pushed too i01.waitTargetPos() Inside i01.finishedGesture()

http://myrobotlab.org/content/movetoblocking-scripting

Worky beter ? 
If yes we have some adjustments to do for existing gestures maybe.

closed 2017-11-16 by moz4r