MyRobotLab/InMoov #117: SRAIXFAILED

hairygael, 2017-10-11:
Hello,
Recently, I cannot ask the robot "what is a * ...." or "How many *...". For some reason it gives me the defaultResponse.
The error seems to be caused by c.myrobotlab.image.DisplayedImage.

Do you have the same issue?

moz4r, 2017-10-11:
Hi Gaël. Yeh same issue.
This sraix depend on a external webservice called ask.pannous.com .
We don't have the "hand" on this.
It seem borked at this time ? ( HTTP ERROR 503 )

I propose this to make it global,solid and sustainable

- Using internal knowledge services, in addition or replacement of pannous ( wikidata + wolfram alpha )
- Cache results

This will be based on french knowledge engine
There may be some work to process some specific hierarchical data (how many ...)
Others ideas welcome

( graph need update )
[image]

hairygael, 2017-10-11:
Hello Anthony, 
The error doesn't come from ask.pannous.com but because something is wrong in our AIML.
If you replace all the AIML files with the original ALICE. aiml, it works just fine and does the request via ask.pannous.com.
Same with the weather request, it works fine with the original ALICE.aiml, but with our AIML it gets SRAIXFAILED.

hairygael, 2017-11-06:
Hello Anthony,
I tried again to figure out because this is annoying.
Many of the request within sraix.aiml do not work anymore.
Could you point me to where (which aiml) is the sraix refers to "WHAT IS THE WEATHER"?

```
<category><pattern>WEATHER FOR *</pattern>
<template><sraix>WHAT IS THE WEATHER <star/></sraix></template>
</category>
```

It's really strange because if I use the original Alice aiml, it works.

moz4r, 2017-11-06:
Hi Gael ! The sraix is the endpoint inside the aiml. 
What it does is a transfert to an hardcoded thing in javaland : 

https://github.com/moz4r/program-ab/blob/master/src/org/alicebot/ab/Sraix.java#L148

When you speak about "original Alice aiml" what is this set, I tested with raw alice set and asked "what is the weather" > the bot awnser me, but it was not "real" forecast : 
Maybe you have an other set I can test to compare, and try to fix this external webservice helper if possible.

https://github.com/Urthen/aiml-en-us-foundation-alice.v1-3/blob/master/atomic.aiml#L6622

moz4r, 2017-11-16:
todo : cook those 3 services Inside English chatbot

http://myrobotlab.org/service/WolframAlpha
http://myrobotlab.org/service/WikiDataFetcher
http://myrobotlab.org/service/OpenWeatherMap

hairygael, 2017-11-16:
That is going to be tedious.
Many SRAIX are still completely functional through ask.pannous, and they
are great answers,
that means the three cooked services shouldn't overlay the working
ask.pannous

Gee, I have no idea how this can be managed...

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

Kakadu31, 2017-12-22:
I found this documentation: https://docs.google.com/document/d/1dVG_B5Sc2x-fi1pN6iJJjfF1bJY6KEFzUqjOb8NsntI/edit#

there it is stated, that the test/guest-key we are using is only available for 1000 searches a day. But there is also stated an email adress at the end where it is probably to ask for a key. Maybe we can try to contact them and get a key for the InMoovs for free? Maybe we can wrap it as the InMoovs can promote their service in the web? I tested the api manually and could never get any result out of it but for news. Thats why I wonder how you could get positive results with the original Alice2 aimls (which i couldn't find and couldn't test it.

hairygael, 2017-12-22:
> document/d/1dVG_B5Sc2x-fi1pN6iJJjfF1bJY6KEFzUqjOb8NsntI/edit#

It sure would be great if the SRAIXFAILED would work again. They use to
work for most of them, but slowly there is less and less pannous request
that work.
Greg and Kevin, proposed to create our own pannous on MRL, it sure would be
even better, but until then it might be a while.
Their TODO list is already so huge.

Thanks for the link, it's good information!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

kwatters, 2018-07-16:
some progress on this... I've updated programab to allow us to provide our own sraix handler.  This means we can intercept the SRAIX requests and add our own implementation based on what ever we want..  This includes things like a call out to open weather map, or even a query to the new solr based robot memory...

hairygael, 2018-07-16:
Great!
Could you point me to those modifications?
We had the openweather map access in the inmoov aiml chatbot but the query
to solr could maybe be added to Manticore.

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

kwatters, 2018-07-16:
Ok,  I just pushed another around of changes..  If you include an OOB tag inside of an SRAIX tag, the oob message will be invoked in a blocking manor and the resulting data from that method call will be converted to a string and added to the ProgramAB response.  Here's a simple example of where we use the method getCurrentBotName on the ProgramAB service to fill in the response template

```
<category>
<pattern>TELL ME THE CURRENT BOT NAME</pattern>
<template>You are talking to **<sraix>**<oob>
      <mrl>
        <service>lloyd</service>
        <method>getCurrentBotName</method>
      </mrl>
    </oob>**</sraix>**</template>
</category>
```

When prompted with the input question  "TELL ME THE CURRENT BOT NAME" 
the above AIML will respond with the teamplte
"You are talking to lloyd"   
the value of "lloyd" is the return value of calling lloyd.getCurrentBotName()

So, any method that returns a string or a number can be used to insert data into the response from ProgramAB now..  which means, you can call out to OpenWeatherMap , or really any service you like to include things in the programab response now.  

This effectively removes the existing SRAIX behavior and overrides it with out own.

kwatters, 2018-07-17:
Yup. sorry I was trying to use the wiki markup to make the tags bold.  :)  I edited my comment to remove the \*\*

moz4r, 2018-09-20:
moved to ogre maybe lot of work Inside aiml files