MyRobotLab/InMoov #118: Error with OpenWeather

hairygael, 2017-10-27:
File "C:/Myrobotlab/myrobotlab.1.0.2596/InMoov/services/H_OpenWeatherMap.py", line 43, in isTheSunShiny
    todayforecast=weather[3].decode("utf-8")
AttributeError: 'NoneType' object has no attribute 'decode'

def isTheSunShiny(townParam=town):
  weather=getRawWeather(townParam)
  
  if weather[1]!=1000 and isChatbotActivated:
    todayforecast=weather[3].decode("utf-8")

moz4r, 2017-11-02:
Hi Gael, fixed ?
( now OpenWeatherMap will warn user about bad api key )
You can get a free one here : http://openweathermap.org/appid

hairygael, 2017-11-03:
Hello Anthony,
I haven't tested yet, but I will try and let you know!

Gael Langevin
Creator of InMoov
InMoov Robot <http://www.inmoov.fr>
@inmoov <http://twitter.com/inmoov>

hairygael, 2017-11-06:
Hello Anthony,
Yes now it's giving the warning for the API key

Great!

closed 2017-11-06 by hairygael